HermanChen/mpp #1: What is the input format for mpi_enc

dalmatele, 2017-09-20:
Hi, I've a problem when using MPP lib.
I can not understand what type of data to input for MPP encode.
When I feed data to mpi task, I always got the packet with same size.

HermanChen, 2017-09-21:
mpi task is a container of MppFrame and MppPacket. You can refer to mpi_enc_test. There is sample of how to attach frame to task and send to mpp. Then output task will carry a packet with its length as its valid data size. The packet size is the fix buffer total length.

dalmatele, 2017-09-21:
@HermanChen:
Thanks for your reply.
When I tried mpi_enc_test, I've given it an input file and I've outputed the value of input buffer and output data of packet but I've had the same problem:
- Input buffer: It alread has had data.
- Output packet: data has had same size with "0 0 0 0 .." value.
And the mpi_enc_test has not reported any problem.
The output video is unuseable.

dalmatele, 2017-09-21:
And this is output I've dumped:

> 0000000 0000 2000 7466 7079 7369 6d6f 0000 0002
0000010 7369 6d6f 7369 326f 7661 3163 706d 3134
0000020 0000 0800 7266 6565 0700 a817 646d 7461
0000030 0000 0000 0000 0000 0000 0000 0000 0000
*
00717d0 0000 c502 6f6d 766f 0000 6c00 766d 6468
00717e0 0000 0000 0000 0000 0000 0000 0000 e803
00717f0 0000 9e15 0100 0000 0001 0000 0000 0000
0071800 0000 0000 0100 0000 0000 0000 0000 0000
*
0071820 0000 0000 0040 0000 0000 0000 0000 0000
0071830 0000 0000 0000 0000 0000 0000 0000 0000
0071840 0000 0200 0000 ef01 7274 6b61 0000 5c00
0071850 6b74 6468 0000 0300 0000 0000 0000 0000
0071860 0000 0100 0000 0000 0000 9e15 0000 0000
0071870 0000 0000 0000 0000 0000 0000 0100 0000
*
0071890 0000 0000 0000 0000 0000 0000 0040 0000
00718a0 3002 0000 4001 0000 0000 2400 6465 7374
00718b0 0000 1c00 6c65 7473 0000 0000 0000 0100
00718c0 0000 9e15 0000 0000 0100 0000 0000 6701
00718d0 646d 6169 0000 2000 646d 6468 0000 0000
00718e0 0000 0000 0000 0000 0000 003c 0100 004c
00718f0 c455 0000 0000 2d00 6468 726c 0000 0000
0071900 0000 0000 6976 6564 0000 0000 0000 0000
0071910 0000 0000 6956 6564 486f 6e61 6c64 7265
0071920 0000 0100 6d12 6e69 0066 0000 7614 686d
0071930 0064 0000 0001 0000 0000 0000 0000 0000
0071940 6424 6e69 0066 0000 641c 6572 0066 0000
0071950 0000 0000 0001 0000 750c 6c72 0020 0000
0071960 0001 0000 73d2 6274 006c 0000 736e 7374
0071970 0064 0000 0000 0000 0001 0000 615e 6376
0071980 0031 0000 0000 0000 0001 0000 0000 0000
0071990 0000 0000 0000 0000 0200 0130 0040 0048
00719a0 0000 0048 0000 0000 0000 0001 0000 0000
00719b0 0000 0000 0000 0000 0000 0000 0000 0000
00719c0 0000 0000 0000 0000 0000 0000 ff18 00ff
00719d0 0000 6108 6376 0043 0000 7318 7474 0073
00719e0 0000 0000 0000 0001 0000 00a6 0200 0000
00719f0 0000 731c 7374 0063 0000 0000 0000 0001
0071a00 0000 0001 0000 00a6 0000 0001 0000 7314
0071a10 7374 007a 0000 0000 0a00 00f0 0000 00a6
0071a20 0000 7314 6374 006f 0000 0000 0000 0001
0071a30 0000 0030 0000 7562 7464 0061 0000 6d5a
0071a40 7465 0061 0000 0000 0000 6821 6c64 0072
0071a50 0000 0000 0000 6d00 6964 6172 7070 006c
0071a60 0000 0000 0000 0000 0000 2d00 6c69 7473
0071a70 0000 2500 74a9 6f6f 0000 1d00 6164 6174
0071a80 0000 0100 0000 0000 614c 6676 3735 372e
0071a90 2e38 3031 0030                         
0071a95

sliver-chen, 2017-09-25:
you should specify input data format in cmd line when use mpi_enc_test

dalmatele, 2017-09-25:
This is my command:
sudo ./mpi_enc_test -i ../../data/small.mp4 -w 560  -h 320 -f 4 -t 7 -d 2 -o output_enc.mp4
But, now I saw in kern.log:
mpp_dev_ioctl:545: unknown mpp ioctl cmd 40086c03
Maybe we have something wrong in kernel?

sliver-chen, 2017-09-25:
@dalmatele 
i see your cmd line
sudo ./mpi_enc_test -i ../../data/small.mp4 -w 560 -h 320 -f 4 -t 7 -d 2 -o output_enc.mp4
1. you specify the input file as small.mp4
so  small.mp4 is a mp4 file? maybe you need to specify as a yuv file accroding to choosed format

dalmatele, 2017-09-25:
@sliver-chen no, the extension is only my file's name. In fact that it's a YUV file. But my output_enc.mp4 is unusable.
As I said above, maybe we have bug in kernel VPU driver.

sliver-chen, 2017-09-25:
@dalmatele 
ok let we get more message from the error log
you can set proc debug option to get more about it.
just do like tihis,
echo  0xffff > /sys/module/vcodec_service/parameters/debug
it is a kernel debug switch,then wo can get more about error

and i suggest you test source file with ffmpeg
using ffmpeg tools to identify it.
ffmpeg -s 560x320 -i small.yuv output_enc.h264
then play it with vlc or elecard tools

Because i can get correctly compress stream by using mpi_enc_test
So maybe we need above operating to check it.

dalmatele, 2017-09-26:
[image]
Sorry, but I can not find this file.
I've attached my list of files in sys directory.

dalmatele, 2017-09-26:
@sliver-chen 
Maybe I've known problem.
In mpp_device.c we had:
`#define VPU_IOC_MAGIC                       'l'

#define VPU_IOC_SET_CLIENT_TYPE             _IOW(VPU_IOC_MAGIC, 1, unsigned long)
#define VPU_IOC_GET_HW_FUSE_STATUS          _IOW(VPU_IOC_MAGIC, 2, unsigned long)
#define VPU_IOC_SET_REG                     _IOW(VPU_IOC_MAGIC, 3, unsigned long)
#define VPU_IOC_GET_REG                     _IOW(VPU_IOC_MAGIC, 4, unsigned long)
#define VPU_IOC_PROBE_IOMMU_STATUS          _IOR(VPU_IOC_MAGIC, 5, unsigned long)
#define VPU_IOC_WRITE(nr, size)             _IOC(_IOC_WRITE, VPU_IOC_MAGIC, (nr), (size))`
But in kernel we defined, mpp_dev_common.h:
`#define MPP_IOC_MAGIC	'l'
#define MPP_IOC_SET_CLIENT_TYPE	_IOW(MPP_IOC_MAGIC, 1, u32)
#define MPP_IOC_GET_HW_FUSE_STATUS	_IOW(MPP_IOC_MAGIC, 2, u32)
#define MPP_IOC_SET_REG	_IOW(MPP_IOC_MAGIC, 3, u32)
#define MPP_IOC_GET_REG	_IOW(MPP_IOC_MAGIC, 4, u32)
#define MPP_IOC_PROBE_IOMMU_STATUS	_IOR(MPP_IOC_MAGIC, 5, u32)`

sliver-chen, 2017-09-26:
@dalmatele 
please send your source file to my email to recurrence problem
my email is sliver.chen@rock-chips.com

dalmatele, 2017-09-26:
@sliver-chen 
I've sent you sample file.
Please check it.
Thank you.

sliver-chen, 2017-09-27:
@dalmatele 
I get the correctly result in my board.
So i suggest you do some check about run environment.
1. check library in board, it need mpp/libmpp.so push to system/lib
2. update code
by the way,you get mpp from this repository,right?

dalmatele, 2017-09-27:
I've used this kernel:
Linux rock64 4.4.77-rockchip-ayufan-93 #1 SMP Tue Aug 29 09:35:00 UTC 2017 aarch64 GNU/Linux
I think it's the newest version.
Can you show me your kernel version?

ethancwchen, 2018-11-21:
firefly@firefly:~/mpp-release/build/linux/aarch64/test$ sudo mpi_enc_test -t 7 -w 640 -h 2
72 -i sintel_480x272_yuv420p.yuv -o t.h264 -f 4
mpi_enc_test: cmd parse result:
mpi_enc_test: input  file name: sintel_480x272_yuv420p.yuv
mpi_enc_test: output file name: t.h264
mpi_enc_test: width      : 640
mpi_enc_test: height     : 272
mpi_enc_test: format     : 4
mpi_enc_test: type       : 7
mpi_enc_test: debug flag : 0
mpi_enc_test: mpi_enc_test start
mpp_rt: NOT found ion allocator
mpp_rt: found drm allocator
mpi_enc_test: mpi_enc_test encoder test start w 640 h 272 type 7
mpi: mpp version: Without VCS info
mpi_enc_test: mpi_enc_test bps 652800 fps 30 gop 60
h264e_api: h264e_config MPP_ENC_SET_RC_CFG bps 652800 [612000 : 693600]
mpi_enc_test: test_mpp_run encoded frame 0 size 0
mpi_enc_test: test_mpp_run encoded frame 1 size 0
mpi_enc_test: test_mpp_run encoded frame 2 size 0
mpi_enc_test: test_mpp_run encoded frame 3 size 0
mpi_enc_test: test_mpp_run encoded frame 4 size 0
mpi_enc_test: test_mpp_run encoded frame 5 size 0
mpi_enc_test: test_mpp_run encoded frame 6 size 0
mpi_enc_test: test_mpp_run encoded frame 7 size 0
mpi_enc_test: test_mpp_run encoded frame 8 size 0
mpi_enc_test: test_mpp_run encoded frame 9 size 0
mpi_enc_test: test_mpp_run encoded frame 10 size 0
mpi_enc_test: test_mpp_run encoded frame 11 size 0
mpi_enc_test: test_mpp_run encoded frame 12 size 0
mpi_enc_test: test_mpp_run encoded frame 13 size 0

closed 2021-05-11 by HermanChen