HermanChen/mpp #2: Problem when using mpi_dec on rock64 board

haidm88, 2017-09-21:
Hi

I'm trying to run mpi_dec_test on rock64 board with follow command:

./mpi_dec_test -i /home/rock64/bunny.h264 -w 1280 -h 720 -t 7 -o /home/rock64/out.yuv

But there're errors and the mpi_dec stop decoding:

mpi_dec_test: cmd parse result:
mpi_dec_test: input  file name: /home/rock64/bunny.h264
mpi_dec_test: output file name: /home/rock64/out.yuv
mpi_dec_test: width      : 1280
mpi_dec_test: height     :  720
mpi_dec_test: type       : 7
mpi_dec_test: debug flag : 0
mpi_dec_test: mpi_dec_test start
mpi_dec_test: input file size 226898646
mpi_dec_test: mpi_dec_test decoder test start w 1280 h 720 type 7
mpi: mpp version: 64b8cdc author: leo.ding [avsd]: add dpb error marking
hal_h264d_api: Assertion vcodec_type & ((0x00000200) | (0x00000001) | (0x00000002)) failed at hal_h264d_init:119
hal_h264d_api: hal_h264d_init hard mode error, value=0
hal_h264d_api: Assertion 0 failed at hal_h264d_init:169
Segmentation fault

Then I tried to use this build https://github.com/rockchip-linux/mpp.  The mpi_dec start decoding, there're an .yuv but it's unuseable .

HermanChen, 2017-09-25:
This mpp version is mainly for Android build. The rockchip-linux/mpp has a lot of patch for linux.
When work on linux it is better to use rockchip-linux version.
We will merge these two branches gradually.

HermanChen, 2017-09-25:
From the log it seems that mpp can not find a hardware device for decoding.

HermanChen, 2017-12-21:
Now we have merged the two branches together now. Please re-check this issue : )

closed 2018-04-09 by HermanChen