HermanChen/mpp #5: Support for 10-bit H264 on Android?

mo123, 2017-12-21:
On Android, mpp don't support 10-bit H264 even if limitation was removed in Kodi 18.
https://github.com/xbmc/xbmc/blob/master/xbmc/cores/VideoPlayer/DVDCodecs/Video/DVDVideoCodecAndroidMediaCodec.cpp#L470-L475
It plays with MediaCodec instead of ffmpeg if limitation is removed but still in 8-bit not 10-bit h264 with lots of color banding.
Is there a solution perhaps as lots of anime videos are in 10-bit and it would be nice to have support for it.

Thanks

HermanChen, 2017-12-21:
RK3399, RK3228 and RK3328 support 10bit H.264 decoding. And I have tested H.264 10bit file 10bitwhite.mkv on RK3328 with latest mpp without any problem.

mo123, 2017-12-21:
Hi

Can you test this video?
https://www.koi-sama.net/files/hi10/[720p][07_REF_L4.0]Sword%20Art%20Online%20BD%20Opening.mkv

And use SPMC I made with 10-bit h264 limitation removed.
https://mega.nz/#!wJtChKIL!Be-EYaiPNnSJelkj1Ep1pbRrIimWaWc6vJeqyknsEdE

Video will switch to 720p@50hz and when the orange picture comes on it will do auto frame-rate switching a second time and play in 1080p on my tv.
The colors look like 8-bit.
But if I play it in MediaCenter app then it plays correctly in 10-bit and video colors are a lot smoother with no color banding or corruption.

If it plays correct, then I maybe need a firmware update or have some patch missing.

Thanks

HermanChen, 2017-12-21:
The 720p SAO Movie can play normally on my 3328 board without any error and display correctly.
But I can not download the SPMC from the link. I just use default VideoPlayer will rkffplayer path.
So it could be a middleware transaction error.

mo123, 2017-12-21:
Here is another download link to try.
http://www.mediafire.com/file/158yodqzqyr2t0n/SPMCapp-arm64-v8a-debug_18_Dec.apk
The video plays fine in MediaCenter app that use rkffplayer on RK3328 or MXPlayer on RK3399 since MXPlayer now has 10-bit support for RK3399.
But Kodi or SPMC that use MediaCodec(mpp) still plays it in 8-bit.
Normal Kodi or SPMC only supports 10-bit H264 by ffmpeg that use 100% CPU but app I linked can support 10-bit H264 by MediaCodec but so far it only plays in 8-bit.

Please check if there is something to make it work in the SPMC linked.

HermanChen, 2017-12-21:
It's better to ask @JamesLinEngineer to help on the player level issue. I am not familiar with this part.

mo123, 2017-12-21:
OK, I sent him a message.
Thanks

closed 2017-12-21 by mo123