HermanChen/mpp #8: mpi_enc_test size = 0

ethancwchen, 2018-11-21:
irefly@firefly:~/mpp-release/build/linux/aarch64/test$ sudo mpi_enc_test -t 7 -w 640 -h 272 -i ds_480x272.yuv -o t.h264 -f 4
mpi_enc_test: cmd parse result:
mpi_enc_test: input  file name: ds_480x272.yuv
mpi_enc_test: output file name: t.h264
mpi_enc_test: width      : 640
mpi_enc_test: height     : 272
mpi_enc_test: format     : 4
mpi_enc_test: type       : 7
mpi_enc_test: debug flag : 0
mpi_enc_test: mpi_enc_test start
mpp_rt: NOT found ion allocator
mpp_rt: found drm allocator
mpi_enc_test: mpi_enc_test encoder test start w 640 h 272 type 7
mpi: mpp version: Without VCS info
mpi_enc_test: mpi_enc_test bps 652800 fps 30 gop 60
h264e_api: h264e_config MPP_ENC_SET_RC_CFG bps 652800 [612000 : 693600]
mpi_enc_test: test_mpp_run encoded frame 0 size 0
mpi_enc_test: test_mpp_run encoded frame 1 size 0
mpi_enc_test: test_mpp_run encoded frame 2 size 0
mpi_enc_test: test_mpp_run encoded frame 3 size 0
mpi_enc_test: test_mpp_run encoded frame 4 size 0
mpi_enc_test: test_mpp_run encoded frame 5 size 0
mpi_enc_test: test_mpp_run encoded frame 6 size 0
mpi_enc_test: test_mpp_run encoded frame 7 size 0
mpi_enc_test: test_mpp_run encoded frame 8 size 0
mpi_enc_test: test_mpp_run encoded frame 9 size 0
mpi_enc_test: test_mpp_run encoded frame 10 size 0

sliver-chen, 2018-12-05:
it seems that you have't a kernel that work,the way you use mpi_enc_test is right and without any error log occur.

HermanChen, 2018-12-06:
Try to run mpp_platform_test and show the log.

closed 2020-09-19 by HermanChen