HermanChen/mpp #9: mpi_enc_test h265e error

ethancwchen, 2018-11-21:
firefly@firefly:~/mpp-release/build/linux/aarch64/test$ sudo mpi_enc_test -t 16777220 -w 640 -h 272 -i ds_480x272.yuv -o t.h264 -f 4  -n 10
mpi_enc_test: cmd parse result:
mpi_enc_test: input  file name: ds_480x272.yuv
mpi_enc_test: output file name: t.h264
mpi_enc_test: width      : 640
mpi_enc_test: height     : 272
mpi_enc_test: format     : 4
mpi_enc_test: type       : 16777220
mpi_enc_test: debug flag : 0
mpi_enc_test: mpi_enc_test start
mpp_rt: NOT found ion allocator
mpp_rt: found drm allocator
mpi_enc_test: mpi_enc_test encoder test start w 640 h 272 type 16777220
mpi: mpp version: Without VCS info
mpp_device: mpp_device_init failed to open device /dev/h265e, errno 12, error msg: Cannot allocate memory
mpi_enc_test: mpi_enc_test bps 652800 fps 30 gop 60
mpp_device: mpp_device_send_reg_with_id ioctl VPU_IOC_WRITE failed ret -1 errno 9 Bad file descriptor
hal_h265e_vepu22: vepu22_set_cfg error: hal_h265e_set_param 0x9 fail
mpp: command 320007 param 0x2d4b11dc ret -1
mpi_enc_test: mpi control enc set codec cfg failed ret -1
mpi_enc_test: mpi_enc_test test mpp setup failed ret -1
mpp_device: mpp_device_deinit invalid negtive file handle,
mpi_enc_test: mpi_enc_test failed ret -1
mpp_buffer: ~MppBufferService cleaning misc group
firefly@firefly:~/mpp-release/build/linux/aarch64/test$

danog, 2019-02-06:
welp got a similar problem here, any solution?

HermanChen, 2020-09-19:
This H.265 encoder is only on RK3328.