HermanChen/mpp #11: mpi_enc_test raw bytes -> jpeg grey image output

qume, 2020-02-26:
I am trying to encode an RGB888 raw image using mpi_enc_test.

Using the following command:
`./mpi_enc_test -i raw.raw -o o.jpg -w 512 -h 512 -f 0x10006 -t 8`

On rockchip3399, with kernel:
`Linux 4.4.210-rockchip64 #1 SMP Sun Jan 19 19:56:09 CET 2020 aarch64 aarch64 aarch64 GNU/Linux`

The raw.raw file has no header, is just packed (not planar) RGB bytes.

The process seems to run correctly:
```mpi_enc_test: cmd parse result:
mpi_enc_test: input  file name: raw.raw
mpi_enc_test: output file name: o.jpg
mpi_enc_test: width      : 512
mpi_enc_test: height     : 512
mpi_enc_test: format     : 0
mpi_enc_test: type       : 8
mpi_enc_test: debug flag : 0
mpi_enc_test: mpi_enc_test start
mpp_rt: NOT found ion allocator
mpp_rt: found drm allocator
mpi_enc_test: mpi_enc_test encoder test start w 512 h 512 type 8
mpi: mpp version: 313156e1 author: sayon.chen [h265e]: Fix compile warn & fixqp case not do rc flow path
mpi_enc_test: mpi_enc_test bps 983040 fps 30 gop 60
jpege_api: jpege: hardware return error status 40
mpi_enc_test: test_mpp_run encoded frame 0 size 33306
mpi_enc_test: test_mpp_run encode max 1 frames
mpi_enc_test: mpi_enc_test success total frame 1 bps 7993440
mpp_buffer: ~MppBufferService cleaning misc group
```

The resulting o.jpg, is apx 50% grey all over. No detail from the original input. Apx every 8x8 pixels is one slightly darker pixel, but this isn't consistent over the whole image. Attached are input and output (note: output isn't a well formed jpeg so may not display properly in the browser).

[image]

[raw.zip](https://github.com/HermanChen/mpp/files/4252823/raw.zip)

There are a significant number of items in dmesg:
```
`
.
.
.[ 4052.311343] rk-vcodec ff650000.vpu_service: reg[179] 00000000
[ 4052.311352] rk-vcodec ff650000.vpu_service: reg[180] 00000000
[ 4052.311361] rk-vcodec ff650000.vpu_service: reg[181] 00000000
[ 4052.311369] rk-vcodec ff650000.vpu_service: reg[182] 00000000
[ 4052.311378] rk-vcodec ff650000.vpu_service: reg[183] 00000000
[ 4052.311566] rk-vcodec ff650000.vpu_service: resetting...
[ 4052.311612] rk-vcodec ff650000.vpu_service: reset done
[ 4052.311655] rk-vcodec ff650000.vpu_service: reset done

```

HermanChen, 2020-02-26:
The jpeg issue is known. We are fixing it.

qume, 2020-02-26:
Thank you. I am building a product which relies on this functionality. Are you able to provide an estimate for when this will be fixed?

Thank you

closed 2020-09-19 by HermanChen