HermanChen/mpp #17: mpp_enc_put_frame 耗时

sunkwei, 2020-09-17:
您好，最近尝试rk3999 android 开发，设备上无 /dev/ion，用 mpi_enc_test -f 4 -t 7 -w 1920 -h 1080 -o /sdcard/test.h264 -n 500 -b 2000000:1600000:2400000 测试，发现 mpp_enc_put_frame 大约耗时 23ms，mpp_enc_get_packet() 耗时很小

  09-17 03:07:48.380  4290  4290 I mpi_enc_test: mpi_enc_test start
  09-17 03:07:48.383  4290  4290 I mpp_rt  : NOT found ion allocator
  09-17 03:07:48.383  4290  4290 I mpp_rt  : found drm allocator
  09-17 03:07:48.393  4290  4290 I mpp_info: mpp version: 46a33b24 author: yandong.lin   2020-09-09 [mpi_enc_test]: add vp8e rc test demo
  09-17 03:07:48.393  4290  4290 I mpi_enc_test: 0x7eacc42000 mpi_enc_test encoder test start w 1920 h 1080 type 7
  09-17 03:07:48.393  4290  4290 I mpi_enc_test: type=1400 audit(0.0:78): avc: denied { read } for name="compatible" dev="sysfs" ino=75 scontext=u:r:shell:s0 tcontext=u:object_r:sysfs:s0 tclass=file permissive=1
  09-17 03:07:48.393  4290  4290 I mpi_enc_test: type=1400 audit(0.0:79): avc: denied { open } for path="/sys/firmware/devicetree/base/compatible" dev="sysfs" ino=75 scontext=u:r:shell:s0 tcontext=u:object_r:sysfs:s0 tclass=file permissive=1
  09-17 03:07:48.399  4290  4291 I h264e_api_v2: MPP_ENC_SET_PREP_CFG w:h [1920:1080] stride [1920:1080]
  09-17 03:07:48.399  4290  4291 I h264e_api_v2: MPP_ENC_SET_RC_CFG bps 2000000 [1600000 : 2400000] fps [30:30] gop 60
  09-17 03:07:48.399  4290  4291 I mpp_enc_v2: send header for set cfg change input/format 
  09-17 03:07:48.393  4290  4290 I mpi_enc_test: type=1400 audit(0.0:80): avc: denied { read write } for name="vpu_service" dev="tmpfs" ino=15873 scontext=u:r:shell:s0 tcontext=u:object_r:video_device:s0 tclass=chr_file permissive=1
  09-17 03:07:48.403  4290  4291 I mpp_enc_v2: mode vbr bps [1600000:2000000:2400000] fps fix [30/1] -> fix [30/1] gop i [60] v [0]
  09-17 03:07:48.393  4290  4290 I mpi_enc_test: type=1400 audit(0.0:81): avc: denied { open } for path="/dev/vpu_service" dev="tmpfs" ino=15873 scontext=u:r:shell:s0 tcontext=u:object_r:video_device:s0 tclass=chr_file permissive=1
  09-17 03:07:48.393  4290  4290 I mpi_enc_test: type=1400 audit(0.0:82): avc: denied { ioctl } for path="/dev/vpu_service" dev="tmpfs" ino=15873 ioctlcmd=6c01 scontext=u:r:shell:s0 tcontext=u:object_r:video_device:s0 tclass=chr_file permissive=1
  09-17 03:07:48.443  4290  4290 I mpi_enc_test: enc: total:0.04013, **put frame:0.04012**, get packet:0.00002
  09-17 03:07:48.445  4290  4290 I mpi_enc_test: 0x7eacc42000 encoded frame 0    size 46213  
  09-17 03:07:48.483  4290  4290 I mpi_enc_test: enc: total:0.03147, **put frame:0.03145**, get packet:0.00002
  09-17 03:07:48.483  4290  4290 I mpi_enc_test: 0x7eacc42000 encoded frame 1    size 12044  
  09-17 03:07:48.513  4290  4290 I mpi_enc_test: enc: total:0.02268, **put frame:0.02266**, get packet:0.00002
  09-17 03:07:48.513  4290  4290 I mpi_enc_test: 0x7eacc42000 encoded frame 2    size 6644   
  09-17 03:07:48.543  4290  4290 I mpi_enc_test: enc: total:0.02286, **put frame:0.02284**, get packet:0.00002

代码中，修改 test_mpp_run() 统计 mpp_enc_put_frame() 和 mpp_enc_get_packet() 的时间。

感觉是数据传输的问题，是否有优化的方案？ 谢谢 :)

HermanChen, 2020-09-17:
put_frame 时间即是编码一帧的时间，因为这一帧的图像在编码过程中都需要被使用，不能释放回来，所以是个阻塞接口，需要等到编码完成才以返回。
这个不是数据传输的问题，是本身实现成这样的。

sunkwei, 2020-09-17:
感谢大佬回复，但rk3399的编码能力这么弱么，本来希望用它支持simulcast 支持至少3路编码，同时解码4,5路呢 ...

HermanChen, 2020-09-17:
解码非常强，4K60fps 很轻松，编码比较弱

sunkwei, 2020-09-17:
好的，谢谢 :)

HermanChen, 2020-09-17:
不客气，问题解决的话麻烦关掉 issue ：）

closed 2020-09-17 by sunkwei