HermanChen/mpp #18: 解码阻塞 ...

sunkwei, 2020-09-17:
您好，我希望给 webrtc 增加 h264 的硬件解码，模仿 mpi_dec_test.c 中的 decode_advanced() 实现，发现 poll(ctx_, MPP_PORT_OUTPUT, MPP_POLL_BLOCK) 时阻塞，没有返回 ...
``` c++
   int mpp_dec(const EncodedImage &img, bool missing)
   {
         MppTask task = 0;
         void *ptr = mpp_buffer_get_ptr(pkt_buf_);  // pkt_buf_ 分配了1M字节空间
         memcpy(ptr, img.data(), img.size());    // 将 EncodedImage 中的压缩数据复制到缓冲.
                                                                      // 数据为IDR，00 00 00 01 67 ....
         mpp_packet_set_pos(packet_, ptr);
         mpp_packet_set_length(packet_, img.size()); 
         rc = mpi_->poll(ctx_, MPP_POPT_INPUT, MPP_POLL_BLOCK);
         rc = mpi_->dequeue(ctx_, MPP_PORT_INPUT, &task);
         mpp_task_meta_set_packet(task, KEY_INPUT_PACKET, &packet_);
         mpp_task_meta_set_frame(task, KEY_OUTPUT_FRAME, &frame_);  // frame_ 已经分配，并绑定到 frm_buf, 大小足够 ...
         mpi_->enqueue(ctx_, MPP_PORT_INPUT, task);
         __android_log_print(6, "mpp", "before poll out ...\n");
         mpi_->poll(ctx_, MPP_PORT_OUTPUT, MPP_POLL_BLOCK);  // << 这一行阻塞了 ...
         ....
```

sunkwei, 2020-09-17:
将 mpi_dec_test.c 中的 
``` c++
cmd->simple = (cmd->type != MPP_VIDEO_CodingMJPEG) ? (1) : (0);
```
改成 cmd->simple = 0;
用 h264 输入测试，也会出现阻塞。 ./mpi_dec_test  -i /sdcard/test.h264 -w 1920 -h 1080 -t 7 也是阻塞在 mpi->poll(ctx, MPP_PORT_OUTPUT, MPP_POLL_BLOCK) 一行，难道这个接口仅仅支持 mjpeg 么？

HermanChen, 2020-09-18:
decode_advanced 中使用的解码器 task 交互模式只在 jpeg 解码时使用，因为 jpeg 解码时用户会要求指定输出图像地址，其他格式没有这种要求，所以这个目前只用在 jpeg 解码上。

sunkwei, 2020-09-18:
好的，谢谢

closed 2020-09-18 by sunkwei