HermanChen/mpp #19: Encoding straight from ISP

jpiat, 2020-09-18:
Hi,

i have issue getting a good FPS out of a video pipeline that encode video acquired through the ISP. The RAW images are acquired through the v4l2 driver for the ISP and then sent to the MPP encoder. Is there a way to do encoding straight from the ISP without the raw video data ever going to userspace ? 

Thanks

PS: why did the mpp repository on linux-rockchip was removed ?

HermanChen, 2020-09-19:
The camera source has been added.

The rockchip-linux mpp is temporally set to private for a while. It will be opened later.
This repo will release the mpp latest commits.

jpiat, 2020-09-21:
I looked at the mpp encoder test with camera input but from what i see its basically using the v4l2 driver for the camera so there is memory copy to user space. Some of the source code of the mpp refers to encoding from isp (with functions left empty), so is there any work in progress to encode straight from ISP ?

HermanChen, 2020-09-21:
No, the buffer is using mmap mode and no copy is required.

jpiat, 2020-09-22:
how can i evaluate frame-rate using this test ?

jpiat, 2020-09-22:
i just did a test using time in inux and the result seems very promising since i encode 3000 frames in 100 seconds which is 30FPS. I'll give it some more test.

jpiat, 2020-09-22:
After looking at the code it seems that example maps the v4l2 buffers to dma so that there is no memcpy involved to get transfer the buffers to the mpp. This works very well but unfortunately it means that applications that needs to process the frame (like gstreamer) will not achieve the max frame rate.

HermanChen, 2020-09-23:
Application can modify the pixel in dma-buf directly or copy to another dma-buf to do the image processing then give it to encoder.

closed 2021-05-11 by HermanChen