HermanChen/mpp #22: 3288 mpi_dec_test error

tongyuhome, 2020-10-12:
Hi, I try to use mpi_dec_test.c in order to get YUV output from JPEG.
But the generated YUV file opens only in green.
The same code can generate correct YUV file on 3399.
The following is the log printed by running mpi_dec_test on 3288:

```
[mpi_dec_test_show_options:902]: cmd parse result:
[mpi_dec_test_show_options:903]: input  file name: ./testImage/testJPEG_640_480.jpeg
[mpi_dec_test_show_options:904]: output file name: ./testImage/output/testJPEG_640_480.yuv
[mpi_dec_test_show_options:905]: config file name: 
[mpi_dec_test_show_options:906]: width      :  640
[mpi_dec_test_show_options:907]: height     :  480
[mpi_dec_test_show_options:908]: type       : 8
[mpi_dec_test_show_options:909]: debug flag : 0
[mpi_dec_test_show_options:910]: max frames : 1
[mpi_dec_test_decode:523]: mpi_dec_test start
[mpi_dec_test_decode:536]: input file size 57512
[MppRuntimeService]-86: found ion allocator
[MppRuntimeService]-91: NOT found drm allocator
[check_sysfs_iommu]-245: vpu_service iommu_enabled 0
[allocator_ion_open]-345: using ion heap ION_HEAP_TYPE_DMA
[mpi_dec_test_decode:628]: 0xb8a33860 mpi_dec_test decoder test start w 640 h 480 type 8
[decode_advanced:476]: 0xb8a33860 decoded frame 0
[decode_advanced:415]: 0xb8a33860 found last packet
[decode_advanced:476]: 0xb8a33860 decoded frame 1
[decode_advanced:480]: 0xb8a33860 found eos frame
[main:944]: test success max memory 0.00 MB
```

[check_sysfs_iommu]-245: vpu_service iommu_enabled 0
[allocator_ion_open]-345: using ion heap ION_HEAP_TYPE_DMA
It seems that there are some problems with ion in the kernel environment

Can you please help me see how to solve this problem?

HermanChen, 2020-10-13:
What is your sdk? Linux or Android?

WainDing, 2020-10-13:
Normally, iommu_enabled should be 1. so, you can follow the code, why is it 0.

tongyuhome, 2020-10-13:
> What is your sdk? Linux or Android?

I compiled with NDK11(also tried 15) on ubuntu16.04, and then used adb tool to run the mpi_dec_test on Android(3288)
I want to know is whether iommu_enabled will affect the use of vpu?

tongyuhome, 2020-10-13:
> Normally, iommu_enabled should be 1. so, you can follow the code, why is it 0.

There is a possibility that ion is not turned on, but there is no way to turn it on on 3288. Any good suggestions?

tongyuhome, 2020-10-13:
> What is your sdk? Linux or Android?

I found that the content of the /proc/device-tree/vpu_service@ff9a0000/iommu_enabled on 3288 is empty，, which leads to vpu_service iommu_enabled 0. Is this some hardware that's not turned on?

HermanChen, 2020-10-13:
> 
> 
> > What is your sdk? Linux or Android?
> 
> I found that the content of the /proc/device-tree/vpu_service@ff9a0000/iommu_enabled on 3288 is empty，, which leads to vpu_service iommu_enabled 0. Is this some hardware that's not turned on?

It may be the problem. Try enable it and test again.

tongyuhome, 2020-10-13:
> > > What is your sdk? Linux or Android?
> > 
> > 
> > I found that the content of the /proc/device-tree/vpu_service@ff9a0000/iommu_enabled on 3288 is empty，, which leads to vpu_service iommu_enabled 0. Is this some hardware that's not turned on?
> 
> It may be the problem. Try enable it and test again.

now, the problem is how to  enable it? This seems to be a hardware issue, do you have any suggestions ?

HermanChen, 2020-10-13:
Change it in dts and try again.

tongyuhome, 2020-10-14:
> Change it in dts and try again.

I'll try it.Thanks!!

closed 2020-11-23 by HermanChen