HermanChen/mpp #24: encode jpeg by 422UYVY format problem

fwanghuan490, 2020-11-04:
Hi,  I download the latest version of mpp, and then I try to use the mpi_enc_test to  encode JPG by a 422UYVY file,  the output jpeg file is not ok,  the jpg file seems half is right，half  has no data. But I encode the same 422UYVY file to encode h264, the output h264 video is ok.

The testing  commad information as below:
./mpi_enc_test -i ../ch0/20201103-234441/20201103-234441-0000.422uyvy -o t2020.jpg -w 2448 -h 2048 -f 10 -t 8

mpp[4797]: mpi_enc_utils: cmd parse result:
mpp[4797]: mpi_enc_utils: input  file name: ../ch0/20201103-234441/20201103-234441-0000.422uyvy
mpp[4797]: mpi_enc_utils: output file name: t2020.jpg
mpp[4797]: mpi_enc_utils: width      : 2448
mpp[4797]: mpi_enc_utils: height     : 2048
mpp[4797]: mpi_enc_utils: format     : 10
mpp[4797]: mpi_enc_utils: type       : 8
mpp[4797]: mpi_enc_test: mpi_enc_test start
mpp[4797]: mpi_enc_test: jpege default encode only one frame. Use -n [num] for rc case
mpp[4797]: mpp_rt: NOT found ion allocator
mpp[4797]: mpp_rt: found drm allocator
mpp[4797]: mpp_info: mpp version: unknown mpp version for missing VCS info
mpp[4797]: mpi_enc_test: 0x558f6fbde0 mpi_enc_test encoder test start w 2448 h 2048 type 8
mpp[4797]: hal_jpege_vepu2_2: hal_jpege_vepu2_set_extra_info other format(10)
mpp[4797]: mpi_enc_test: test_mpp_run packet flag:0
mpp[4797]: mpi_enc_test: test_mpp_run FFFFFFFFFFFFFFFFFFFFF is_intra:1
mpp[4797]: mpi_enc_test: 0x558f6fbde0 encoded frame 0    size 202505
mpp[4797]: mpi_enc_test: 0x558f6fbde0 encode max 1 frames
mpp[4797]: mpi_enc_test: 0x558f6fbde0 mpi_enc_test success total frame 1 bps 48601200

So, what's wrong with the JPEG encoding by 422UYVY format ?

HermanChen, 2020-11-04:
Which chip you are using? Try run mpp_platform_test and show the result.
用的什么芯片平台？跑下 mpp_platform_test 看下结果

fwanghuan490, 2020-11-04:
用的RK3399，mpp_paltform_test执行如下：

root@NanoPC-T4:~/test/rk3399extlib/mpp-develop/build/linux/aarch64/osal/test# ./mpp_platform_test
mpp[6902]: mpp_plat_test: chip name: friendlyelec,nanopc-t4 rockchip,rk3399
mpp[6902]: mpp_plat_test:
mpp[6902]: mpp_plat_test: chip vcodec type 00040202
mpp[6902]: mpp_plat_test: found vpu2 codec
mpp[6902]: mpp_plat_test: found rkvdec decoder
mpp[6902]: mpp_plat_test: found vpu2 encoder
mpp[6902]: mpp_plat_test:
mpp[6902]: mpp_plat_test: start probing decoder device name:
mpp[6902]: mpp_plat_test: H.264 decoder: /dev/rkvdec
mpp[6902]: mpp_plat_test: H.265 decoder: /dev/rkvdec
mpp[6902]: mpp_plat_test: MJPEG decoder: /dev/vpu_service
mpp[6902]: mpp_plat_test: VP9   decoder: /dev/rkvdec
mpp[6902]: mpp_plat_test: avs   decoder: /dev/vpu_service
mpp[6902]: mpp_plat_test:
mpp[6902]: mpp_plat_test: start probing encoder device name:
mpp[6902]: mpp_plat_test: H.264 encoder: /dev/vpu_service
mpp[6902]: mpp_plat_test: H.265 encoder: /dev/vpu_service
mpp[6902]: mpp_plat_test: MJPEG encoder: /dev/vpu_service
mpp[6902]: mpp_plat_test: mpp platform test done

HermanChen, 2020-11-04:
了解，这个应该是 stride 配置问题，我们也确认一下
你的 case 是读文件的流程，你那边可以对比下 read_image 和 fill_image 两个函数看一下，主要是 stride 问题

fwanghuan490, 2020-11-04:
请问stride该怎么配呢？
我如果用fill_image输出的jpg图片也是不对的，上半部分显示似乎是切割成了好几个部分的五颜六色的斜块格子，下半部分图片仍然是全绿色。
这是我执行fill_image的打印：
root@NanoPC-T4:~/test/rk3399extlib/mpp-develop/build/linux/aarch64/test# ./mpi_enc_test -o t2021.jpg -w 1920 -h 1080 -f 10 -t 8
mpp[13265]: mpi_enc_utils: cmd parse result:
mpp[13265]: mpi_enc_utils: input  file name: (null)
mpp[13265]: mpi_enc_utils: output file name: t2021.jpg
mpp[13265]: mpi_enc_utils: width      : 1920
mpp[13265]: mpi_enc_utils: height     : 1080
mpp[13265]: mpi_enc_utils: format     : 10
mpp[13265]: mpi_enc_utils: type       : 8
mpp[13265]: mpi_enc_test: mpi_enc_test start
mpp[13265]: mpi_enc_test: jpege default encode only one frame. Use -n [num] for rc case
mpp[13265]: mpp_rt: NOT found ion allocator
mpp[13265]: mpp_rt: found drm allocator
mpp[13265]: mpp_info: mpp version: unknown mpp version for missing VCS info
mpp[13265]: mpi_enc_test: 0x55cb3f2b60 mpi_enc_test encoder test start w 1920 h 1080 type 8
mpp[13265]: mpi_enc_test: AAAAAA fill_image w:1920, h:1080, hor_stride:3840, CTX:0x0
mpp[13265]: utils: BBBBBB fill_image MPP_FMT_YUV422_UYVY w:1920, h:1080, hor_stride:3840
mpp[13265]: hal_jpege_vepu2_2: hal_jpege_vepu2_set_extra_info other format(10)
mpp[13265]: mpi_enc_test: 0x55cb3f2b60 encoded frame 0    size 138583
mpp[13265]: mpi_enc_test: 0x55cb3f2b60 encode max 1 frames
mpp[13265]: mpi_enc_test: 0x55cb3f2b60 mpi_enc_test success total frame 1 bps 33259920

其中mpp[13265]: mpi_enc_test: AAAAAA和mpp[13265]: utils: BBBBBB这两条打印是我自己在demo里加的打印stride值的。

麻烦指点一下该如何配置正确的stride值。

HermanChen, 2020-11-04:
用 -u 选项来配置 hor_stride，分别加 -u 1920 和 -u 3840 两种配置来测试看下

fwanghuan490, 2020-11-04:
加-u 1920和-u 3840都还是不对啊，现在都是用fill_image来编码试了：

./mpi_enc_test -o t2021.jpg -w 1920 -h 1080 -f 10 -t 8 -u 1920

./mpi_enc_test -o t2021.jpg -w 1920 -h 1080 -f 10 -t 8 -u 3840

HermanChen, 2020-11-04:
代码己更新，应该修正了这个问题，麻烦再测试一下

fwanghuan490, 2020-11-04:
感谢，代码更新后，经测试422编码JPG和H264都可以了。
另外，我用RGB888格式编码H264时，发现打印提示不支持这个格式，但是mpp_enc_test的demo又是支持，这个是咋回事呢？

mpp[32292]: mpi_enc_test: 0x55aa2a6e00 mpi_enc_test encoder test start w 1920 h 1080 type 7
mpp[32292]: h264e_api_v2: MPP_ENC_SET_PREP_CFG w:h [1920:1080] stride [5760:1080]
mpp[32292]: h264e_api_v2: MPP_ENC_SET_RC_CFG bps 7776000 [486000 : 8262000] fps [30:30] gop 60
mpp[32292]: mpp_enc_v2: send header for set cfg change input/format
mpp[32292]: mpp_enc_v2: mode vbr bps [486000:7776000:8262000] fps fix [30/1] -> fix [30/1] gop i [60] v [0]
mpp[32292]: vepu_common: get_vepu_fmt unsupport frame format 10006
mpp[32292]: hal_h264e_vepu_v2: vepu do not support input frame format 65542
mpp[32292]: vepu_common: get_vepu_pixel_stride invalid fmt 65542

HermanChen, 2020-11-05:
24bit 的 RGB888 对于硬件来说是不支持的，mpi_enc_test 是支持生成这样的数据格式，所以给硬件之后，硬件上报错误是正常的

closed 2020-11-05 by HermanChen