HermanChen/mpp #25: rk3399上用mpp进行Jpeg编码时如何调整压缩率

fwanghuan490, 2020-11-11:
你好，如题所述rk3399上用mppJpeg编码时如何调整压缩率？我现在用mpp编码yuv422格式的输入文件后，生成的jpg文件能正常显示，但是还是有些大，mpp中调整哪个参数可以提高压缩比？

HermanChen, 2020-11-11:
调节 jpeg:q_factor 这个参数，在 mpi_enc_test 里可以找到

fwanghuan490, 2020-11-11:
这个参数改变我也测过，jpeg:q_factor 设置为1，10，50都没变化

qvoid, 2020-11-11:
rc mode配置成什么，如果是fix qp模式，配置q_factor会不起作用

fwanghuan490, 2020-11-11:
目前配置的rc_cfg->rc_mode模式是1，即MPP_ENC_RC_MODE_CBR

qvoid, 2020-11-11:
我这边用mpi_enc_test测试，修改jpeg:q_factor测试，编码出的图片质量有变化，确实有生效。测试的值是10 和80
执行下如下的设置环境变量的cmd，然后再用mpi_enc_test测试
```
setprop hal_jpege_debug 0x10
```
看下logcat中类似这样log的打印出来的qfactor再更改前后是否有变化：
```
01-03 11:28:47.879: I/hal_jpege_vepu2_2(2094): use qfactor=10, rc_info->quality_target=90
``

fwanghuan490, 2020-11-11:
用mpi_enc_test编码出来确实比较小，但是我的程序是多线程，是基于mpi_enc_multi_test这个demo来写的，似乎mpi_enc_multi_test不行

qvoid, 2020-11-11:
确认下 log中打印出来的q_factor是否有变化，如果没有，那就是配置更新的问题，检查下流程

fwanghuan490, 2020-11-11:
我现在按照mpi_enc_test的流程重新写了一下更新到我自己的工程里面去，这次q_factor确实起作用了，但还是有点问题，不知道该怎么解决。因为我是连续Jpeg编码，不像mpi_enc_test那样只编码一帧，发现编码第一帧时大小确实是对的（同样的输入源和mpi_enc_test编码出来的第一帧一样大），但从第2帧还是编出来的Jpeg文件大小就一直在递减，一直减小到80K左右就稳定不再变小了，日志如下：
[ats:2020-11-11 23:10:36 @../src/media/Public/JpegSequencesRkMgr.cpp:FrameWriteFile#75][DEBUG] enc:138,wirte:13 ms,all:151 ms, inputLen[10027008], outputLen[543264]
[ats:2020-11-11 23:10:36 @../src/media/Public/JpegSequencesRkMgr.cpp:FrameWriteFile#75][DEBUG] enc:122,wirte:22 ms,all:144 ms, inputLen[10027008], outputLen[343135]
[ats:2020-11-11 23:10:36 @../src/media/Public/JpegSequencesRkMgr.cpp:FrameWriteFile#75][DEBUG] enc:84,wirte:6 ms,all:90 ms, inputLen[10027008], outputLen[258834]
[ats:2020-11-11 23:10:37 @../src/media/Public/JpegSequencesRkMgr.cpp:FrameWriteFile#75][DEBUG] enc:77,wirte:6 ms,all:83 ms, inputLen[10027008], outputLen[236485]
[ats:2020-11-11 23:10:37 @../src/media/Public/JpegSequencesRkMgr.cpp:FrameWriteFile#75][DEBUG] enc:59,wirte:5 ms,all:64 ms, inputLen[10027008], outputLen[201726]
[ats:2020-11-11 23:10:37 @../src/media/Public/JpegSequencesRkMgr.cpp:FrameWriteFile#75][DEBUG] enc:74,wirte:4 ms,all:78 ms, inputLen[10027008], outputLen[200175]
[ats:2020-11-11 23:10:37 @../src/media/Public/JpegSequencesRkMgr.cpp:FrameWriteFile#75][DEBUG] enc:59,wirte:4 ms,all:63 ms, inputLen[10027008], outputLen[184821]
[ats:2020-11-11 23:10:37 @../src/media/Public/JpegSequencesRkMgr.cpp:FrameWriteFile#75][DEBUG] enc:63,wirte:5 ms,all:68 ms, inputLen[10027008], outputLen[176354]
[ats:2020-11-11 23:10:37 @../src/media/Public/JpegSequencesRkMgr.cpp:FrameWriteFile#75][DEBUG] enc:61,wirte:4 ms,all:65 ms, inputLen[10027008], outputLen[168472]
[ats:2020-11-11 23:10:37 @../src/media/Public/JpegSequencesRkMgr.cpp:FrameWriteFile#75][DEBUG] enc:59,wirte:5 ms,all:64 ms, inputLen[10027008], outputLen[165253]
[ats:2020-11-11 23:10:37 @../src/media/Public/JpegSequencesRkMgr.cpp:FrameWriteFile#75][DEBUG] enc:67,wirte:4 ms,all:71 ms, inputLen[10027008], outputLen[156797]
[ats:2020-11-11 23:10:37 @../src/media/Public/JpegSequencesRkMgr.cpp:FrameWriteFile#75][DEBUG] enc:61,wirte:3 ms,all:64 ms, inputLen[10027008], outputLen[153756]
[ats:2020-11-11 23:10:37 @../src/media/Public/JpegSequencesRkMgr.cpp:FrameWriteFile#75][DEBUG] enc:60,wirte:4 ms,all:64 ms, inputLen[10027008], outputLen[152195]
[ats:2020-11-11 23:10:37 @../src/media/Public/JpegSequencesRkMgr.cpp:FrameWriteFile#75][DEBUG] enc:60,wirte:3 ms,all:63 ms, inputLen[10027008], outputLen[149126]
[ats:2020-11-11 23:10:37 @../src/media/Public/JpegSequencesRkMgr.cpp:FrameWriteFile#75][DEBUG] enc:64,wirte:3 ms,all:67 ms, inputLen[10027008], outputLen[144496]
[ats:2020-11-11 23:10:37 @../src/media/Public/JpegSequencesRkMgr.cpp:FrameWriteFile#75][DEBUG] enc:61,wirte:3 ms,all:64 ms, inputLen[10027008], outputLen[142316]
[ats:2020-11-11 23:10:37 @../src/media/Public/JpegSequencesRkMgr.cpp:FrameWriteFile#75][DEBUG] enc:60,wirte:3 ms,all:63 ms, inputLen[10027008], outputLen[138354]
[ats:2020-11-11 23:10:38 @../src/media/Public/JpegSequencesRkMgr.cpp:FrameWriteFile#75][DEBUG] enc:62,wirte:3 ms,all:65 ms, inputLen[10027008], outputLen[136408]
[ats:2020-11-11 23:10:38 @../src/media/Public/JpegSequencesRkMgr.cpp:FrameWriteFile#75][DEBUG] enc:62,wirte:3 ms,all:65 ms, inputLen[10027008], outputLen[132826]
[ats:2020-11-11 23:10:38 @../src/media/Public/JpegSequencesRkMgr.cpp:FrameWriteFile#75][DEBUG] enc:59,wirte:3 ms,all:62 ms, inputLen[10027008], outputLen[130759]
[ats:2020-11-11 23:10:38 @../src/media/Public/JpegSequencesRkMgr.cpp:FrameWriteFile#75][DEBUG] enc:59,wirte:3 ms,all:62 ms, inputLen[10027008], outputLen[126911]
[ats:2020-11-11 23:10:38 @../src/media/Public/JpegSequencesRkMgr.cpp:FrameWriteFile#75][DEBUG] enc:64,wirte:3 ms,all:67 ms, inputLen[10027008], outputLen[124794]
[ats:2020-11-11 23:10:38 @../src/media/Public/JpegSequencesRkMgr.cpp:FrameWriteFile#75][DEBUG] enc:61,wirte:3 ms,all:64 ms, inputLen[10027008], outputLen[120902]
[ats:2020-11-11 23:10:38 @../src/media/Public/JpegSequencesRkMgr.cpp:FrameWriteFile#75][DEBUG] enc:60,wirte:3 ms,all:63 ms, inputLen[10027008], outputLen[117993]
[ats:2020-11-11 23:10:38 @../src/media/Public/JpegSequencesRkMgr.cpp:FrameWriteFile#75][DEBUG] enc:59,wirte:2 ms,all:61 ms, inputLen[10027008], outputLen[114293]
[ats:2020-11-11 23:10:38 @../src/media/Public/JpegSequencesRkMgr.cpp:FrameWriteFile#75][DEBUG] enc:63,wirte:3 ms,all:66 ms, inputLen[10027008], outputLen[112608]
[ats:2020-11-11 23:10:38 @../src/media/Public/JpegSequencesRkMgr.cpp:FrameWriteFile#75][DEBUG] enc:61,wirte:2 ms,all:63 ms, inputLen[10027008], outputLen[109103]
[ats:2020-11-11 23:10:38 @../src/media/Public/JpegSequencesRkMgr.cpp:FrameWriteFile#75][DEBUG] enc:59,wirte:2 ms,all:61 ms, inputLen[10027008], outputLen[108266]
[ats:2020-11-11 23:10:38 @../src/media/Public/JpegSequencesRkMgr.cpp:FrameWriteFile#75][DEBUG] enc:60,wirte:2 ms,all:62 ms, inputLen[10027008], outputLen[104185]
[ats:2020-11-11 23:10:38 @../src/media/Public/JpegSequencesRkMgr.cpp:FrameWriteFile#75][DEBUG] enc:61,wirte:3 ms,all:64 ms, inputLen[10027008], outputLen[102035]
[ats:2020-11-11 23:10:38 @../src/media/Public/JpegSequencesRkMgr.cpp:FrameWriteFile#75][DEBUG] enc:61,wirte:2 ms,all:63 ms, inputLen[10027008], outputLen[97709]
[ats:2020-11-11 23:10:38 @../src/media/Public/JpegSequencesRkMgr.cpp:FrameWriteFile#75][DEBUG] enc:59,wirte:2 ms,all:61 ms, inputLen[10027008], outputLen[95825]
[... cut ...]
[ats:2020-11-11 23:10:39 @../src/media/Public/JpegSequencesRkMgr.cpp:FrameWriteFile#75][DEBUG] enc:59,wirte:2 ms,all:61 ms, inputLen[10027008], outputLen[85569]
[ats:2020-11-11 23:10:39 @../src/media/Public/JpegSequencesRkMgr.cpp:FrameWriteFile#75][DEBUG] enc:63,wirte:2 ms,all:65 ms, inputLen[10027008], outputLen[83921]
[ats:2020-11-11 23:10:39 @../src/media/Public/JpegSequencesRkMgr.cpp:FrameWriteFile#75][DEBUG] enc:61,wirte:2 ms,all:63 ms, inputLen[10027008], outputLen[82721]

这个里面,outputLen就是我mpp编码后输出的jpeg文件大小，q_factor我设置的90，然后第一帧输出是530K左右，这是对的，然后后面的第2帧，3帧开始就一直在递减了，直到稳定在80K左右大小的输出，然后编码出来的80K左右Jpeg图像质量就比较不理想了，颜色失真就比较严重了。
这个问题是调什么参数来解决呢？

qvoid, 2020-11-12:
因为你这里配置成motion jpeg的CBR 码控模式，会根据设置的比特率动态调整码率。设置成固定QP模式，则q_factor是会保持不变的

fwanghuan490, 2020-11-12:
好的，用fix_qp可以了，谢谢！

closed 2020-11-12 by qvoid