HermanChen/mpp #27: mpp连续解码Jpeg问题

fwanghuan490, 2020-11-13:
Hi，rk3399 linux平台使用最新版本的mpp进行jpeg解码输出yuv420sp格式的jpg文件，发现在连续解码时，头两帧都是正常输出，但是从第3帧开始开始出现异常打印错误，而且从第4帧开始输出的文件全部都是和第3帧的jpg输出文件一样，出错日志如下：
mpp[22973]: mpi_dec_test: mpi_dec_test_decode the 1 decode
mpp[22973]: mpi_dec_test: decode_advanced 0x5592c2c0a0 decoded frame 0
mpp[22973]: mpi_dec_test: mpi_dec_test_decode the 2 decode
mpp[22973]: mpi_dec_test: decode_advanced 0x5592c2c0a0 decoded frame 1
mpp[22973]: mpi_dec_test: mpi_dec_test_decode the 3 decode
mpp[22973]: mpi_dec_test: decode_advanced 0x5592c2c0a0 decoded frame 2
mpp[22973]: mpi_dec_test: mpi_dec_test_decode the 4 decode
mpp[22973]: HAL_JPEG_VDPU2: hal_jpegd_vdpu2_wait IRQ TIMEOUT!
mpp[22973]: mpi_dec_test: decode_advanced 0x5592c2c0a0 decoded frame 3
mpp[22973]: mpi_dec_test: mpi_dec_test_decode the 5 decode
mpp[22973]: HAL_JPEG_VDPU2: hal_jpegd_vdpu2_wait IRQ TIMEOUT!
mpp[22973]: mpi_dec_test: decode_advanced 0x5592c2c0a0 decoded frame 4
mpp[22973]: mpi_dec_test: mpi_dec_test_decode the final decode over

就是从第3帧开始，一旦出现打印HAL_JPEG_VDPU2: hal_jpegd_vdpu2_wait IRQ TIMEOUT!后，解码输出的文件就不对了，请问这个错误该怎么解决。

fwanghuan490, 2020-11-13:
纠正一下，应该是从第4帧解码开始出现的错误打印

fwanghuan490, 2020-11-18:
这个问题请问大概是什么原因造成的？

HermanChen, 2020-11-23:
麻烦提供一下测试文件？

fwanghuan490, 2020-11-23:
好的，就连续的解码这个jpg文件就可以了：
百度网盘链接：https://pan.baidu.com/s/1LbM2zQX4k95EYlIvIZvLoA 
提取码：r4z0 

为了测试连续解码，我把demo中的代码增加了几行：
在mpi_dec_test.c文件中的mpi_dec_test_decode()函数中大概第677行中的while (!data.eos)里面，新增了多次调用decode_advanced(),具体如下：
while (!data.eos) {
            mpp_err_f("the 1 decode\n");
            decode_advanced(&data);
            fseek(data.fp_input, 0L, SEEK_END);
            file_size = ftell(data.fp_input);
            rewind(data.fp_input);
            mpp_err_f("the 2 decode\n");
            decode_advanced(&data);
            fseek(data.fp_input, 0L, SEEK_END);
            file_size = ftell(data.fp_input);
            rewind(data.fp_input);
            mpp_err_f("the 3 decode\n");
            decode_advanced(&data);
            fseek(data.fp_input, 0L, SEEK_END);
            file_size = ftell(data.fp_input);
            rewind(data.fp_input);
            mpp_err_f("the 4 decode\n");
            decode_advanced(&data);
            fseek(data.fp_input, 0L, SEEK_END);
            file_size = ftell(data.fp_input);
            rewind(data.fp_input);
            mpp_err_f("the 5 decode\n");
            decode_advanced(&data);
            mpp_err_f("the final decode over\n");
        }

FumasterLin, 2020-12-01:
Hi：
你提供的百度网盘链接失效了，麻烦重新提供下

fwanghuan490, 2020-12-01:
你好，Jpg文件新的百度网盘链接如下：

链接：https://pan.baidu.com/s/1Mpg0URLYLUwBa0c5WsYL1g 
提取码：16lw

FumasterLin, 2020-12-01:
Hi：
提供下你本地最新mpp的版本信息？
另外，要测试连续解同一帧jpeg的话，请合入如下修改，再进一步测试下看看。

用以下命令测试（加入-n 参数 指定重复编多少帧）：
mpi_dec_test -i t2020.jpg -w 2448 -h 2048 -t 8 -n 10 -o 2448x2048.yuv

diff --git a/test/mpi_dec_test.c b/test/mpi_dec_test.c
index f8c9557c..cd5f0d22 100644
--- a/test/mpi_dec_test.c
+++ b/test/mpi_dec_test.c
@@ -473,6 +473,14 @@ static int decode_advanced(MpiDecLoopData *data)
                 dump_mpp_frame_to_file(frame, data->fp_output);
 
             mpp_log("%p decoded frame %d\n", ctx, data->frame_count);
+
+            if (data->frame_num > 0 && data->frame_count < data->frame_num) {
+                data->eos = 0;
+                clearerr(data->fp_input);
+                rewind(data->fp_input);
+            }else{
+                data->eos = 1;
+            }
             data->frame_count++;
 
             if (mpp_frame_get_eos(frame_out))