HermanChen/mpp #28: Missing header files

Nybon, 2020-11-13:
Hello.

I'm building mpp into a Qt program and facing difficulties while importing the necessary header files. I'm building my program on a Linux distribution and the media library struggles to find header files located in other folders of the library. 

Here is a picture illustrating what I am talking about. 
[image]
 My code finds every header file, but the header files can't find other header files that are not located in the same folder. As shown in the picture, _mpp-develop\mpp\base\mpp_buf_slot.cpp_  is looking for _mpp_log.h_, but it crashes since it is not in the same folder. 

So, what is actually the correct way of handling the library files? Where do I place the folders from this API?

HermanChen, 2020-11-14:
The cmake script handled the include directories by include_directories in CMakeLists.txt

Nybon, 2020-11-15:
I changed my build system to cmake and here is my CMakeLists.txt file. 
[image]
What should I add to my CMakeLists.txt?

HermanChen, 2020-11-16:
include_directories(your_directory)

Nybon, 2020-11-16:
[image]
I'm now able to find all the headers, but I am not able to use the functions. What should I write in my CMakeLists.txt to fix this?

HermanChen, 2020-11-17:
You also need to add more source code.
The mpp_packet is in mpp/base/

Nybon, 2020-11-17:
This example https://github.com/WainDing/mpp_linux_cpp/blob/master/CMakeLists.txt imports the whole library with target_link_libraries(mpp_linux_demo mpp pthread drm). 

Can I use something similar in my situation?

I added these lines to my CMakeLists.txt file:
include_directories(inc)
include_directories(utils)
include_directories(osal/inc)
include_directories(mpp/inc)
add_subdirectory(mpp)
[image]

HermanChen, 2020-11-18:
Yes, it is right.

Nybon, 2020-11-18:
I am using target_link_libraries(mpp_linux_demo mpp pthread drm).

I have included the following directories:
include_directories(inc)
include_directories(utils)
include_directories(osal/inc)
include_directories(mpp/inc)
include_directories(mpp/hal/inc)
add_subdirectory(mpp)

And I am still getting the error messages shown in the picture above.

HermanChen, 2020-11-19:
Sub-directory should be also added. The sub-diretory CMakeLists.txt should be recursively included.

Nybon, 2020-11-19:
I have added:
add_subdirectory(mpp)
add_subdirectory(osal)
add_subdirectory(utils)
add_subdirectory(test)
add_subdirectory(mpp-develop)
add_subdirectory(hal)
add_subdirectory(base)
add_subdirectory(codec)
add_subdirectory(mpp/legacy)
add_subdirectory(mpp/hal)
add_subdirectory(mpp)
add_subdirectory(osal)
add_subdirectory(test)
add_subdirectory(utils)
add_subdirectory(mpp/hal)
add_subdirectory(mpp/base)
add_subdirectory(mpp/codec)
add_subdirectory(mpp/vproc)
add_subdirectory(osal/test)
add_subdirectory(mpp/legacy)
add_subdirectory(mpp/hal/vpu)
add_subdirectory(mpp/codec/rc)
add_subdirectory(mpp/base/test)
add_subdirectory(mpp/codec/dec)
add_subdirectory(mpp/codec/enc)
add_subdirectory(mpp/hal/dummy)
add_subdirectory(mpp/hal/rkenc)
add_subdirectory(mpp/vproc/iep)
add_subdirectory(mpp/vproc/rga)
add_subdirectory(mpp/hal/common)
add_subdirectory(mpp/hal/vpu/common)
add_subdirectory(mpp/vproc/iep/test)

But nothing new happens.

HermanChen, 2020-11-23:
add_subdirectory is to add the CMakeLists.txt in the subdirectory. Use include_directories also to add include directory.

closed 2021-02-07 by HermanChen

EhsanVahab, 2022-02-09:
> I have added: add_subdirectory(mpp) add_subdirectory(osal) add_subdirectory(utils) add_subdirectory(test) add_subdirectory(mpp-develop) add_subdirectory(hal) add_subdirectory(base) [...]

Dear @Nybon
did you solve the problem?
could you please share your experience?

Nybon, 2022-02-10:
> Dear @Nybon did you solve the problem? could you please share your experience?

I never got it to work, but I found another solution. The project contains samples that will run fine. Try editing the samples instead of building the whole thing from scrath. It may be a bit messy, but it will at least work. Please share a better way to do this if you find one. Remember to keep the same file positions!

EhsanVahab, 2022-02-10:
> I never got it to work, but I found another solution. The project contains samples that will run fine. Try editing the samples instead of building the whole thing from scrath. It may be a bit messy, [...]

I'm trying to make a simple and clear C++ example.
let you know if I do.