HermanChen/mpp #29: JPEG 编码异常

Justa1986, 2020-11-16:
主仓库的，jpeg编码质量过差
开发分支仓库（本仓库），jpeg编码完全异常掉  

Android版本：9.0
平台：RK3288W

测试指令：
adb shell /app/bin/mpi_enc_test -i /tmp/1.yuv -o /tmp/1.jpg -t 8 -f 0 -w 1920 -h 1080 -u 1920 -v 1080 
原图YUV：
[1.yuv.zip](https://github.com/HermanChen/mpp/files/5543653/1.yuv.zip)

编码异常图片：[image]
编码质量过差（主仓库），注意脸部分颜色异常：
[image]

HermanChen, 2020-11-16:
运行看下 mpp_platform_test 的 log
可以不用提供 -u 和 -v 参数，内部会自适应处理

Justa1986, 2020-11-17:
# mpp_platform_test 运行log如下
11-16 21:55:45.754  8147  8147 I mpp_plat_test: chip name: rockchip,rk3288w
11-16 21:55:45.754  8147  8147 I mpp_plat_test: 
11-16 21:55:45.754  8147  8147 I mpp_plat_test: chip vcodec type 00020101
11-16 21:55:45.754  8147  8147 I mpp_plat_test: found vpu1 codec
11-16 21:55:45.754  8147  8147 I mpp_plat_test: found rk hevc decoder
11-16 21:55:45.754  8147  8147 I mpp_plat_test: 
11-16 21:55:45.754  8147  8147 I mpp_plat_test: start probing decoder device name:
11-16 21:55:45.754  8147  8147 I mpp_plat_test: H.264 decoder: /dev/vpu-service
11-16 21:55:45.754  8147  8147 I mpp_plat_test: H.265 decoder: /dev/hevc-service
11-16 21:55:45.754  8147  8147 I mpp_plat_test: MJPEG decoder: /dev/vpu-service
11-16 21:55:45.754  8147  8147 I mpp_plat_test: VP9   decoder: /dev/vpu-service
11-16 21:55:45.754  8147  8147 I mpp_plat_test: avs   decoder: /dev/vpu-service
11-16 21:55:45.754  8147  8147 I mpp_plat_test: 
11-16 21:55:45.754  8147  8147 I mpp_plat_test: start probing encoder device name:
11-16 21:55:45.754  8147  8147 I mpp_plat_test: H.264 encoder: /dev/vpu-service
11-16 21:55:45.755  8147  8147 I mpp_plat_test: H.265 encoder: /dev/vpu-service
11-16 21:55:45.755  8147  8147 I mpp_plat_test: MJPEG encoder: /dev/vpu-service
11-16 21:55:45.755  8147  8147 I mpp_plat_test: mpp platform test done

# 移除-u -v 参数，效果一样还是异常

qvoid, 2020-11-17:
更新到最新的develop分支代码试试看,我这边测试正常

Justa1986, 2020-11-18:
已更新到develop分支，测试效果还是一样，可以提供远程adb环境，帮忙看下？

commit d1e47f8137c8c568fe47c42fa9723b88aa016d8f (HEAD -> develop, origin/develop, origin/HEAD)
Author: shine.liu <shine.liu@rock-chips.com>
Date:   Tue Nov 17 09:27:32 2020 +0800

    [hal_h263d]: Remove unused code in hal_api_h263d
    
    Change-Id: Ia635841b353849033003c02548550e16f4dbe8d7
    Signed-off-by: shine.liu <shine.liu@rock-chips.com>

HermanChen, 2020-11-18:
同样芯片的其他板子也一样的问题么？

Justa1986, 2020-11-18:
嗯，也是同样问题，我试了几块现象都一致

qvoid, 2020-11-18:
执行如下命令后，抓下logcat 和kernel的日志
```
setprop jpege_debug 0xff
echo 0xffff > /sys/module/rk_vcodec/parameters/debug
```

Justa1986, 2020-11-18:
设置之后，编码正常了，但是画质很差，log见附件
[image]

[kernel.txt](https://github.com/HermanChen/mpp/files/5558272/kernel.txt)
[logcat.txt](https://github.com/HermanChen/mpp/files/5558273/logcat.txt)

Justa1986, 2020-11-18:
setprop jpege_debug 0xff
确认这个改动，会影响到编码结果，这个多了几条的logcat消息的输出，会有正常图像出来。

如果是 setprop jpege_debug 0x0
则回到异常的编码效果

qvoid, 2020-11-18:
默认q_factor 80，你这个怎么会是98，而且实际编码写来，q_factor变得很小。是不是改动了代码？

Justa1986, 2020-11-18:
画质有改过，调整成98，我改回去，重新抓个log，跟这个实际理论无关

画质80，未对源码修改logcat详见附件
[kernel.txt](https://github.com/HermanChen/mpp/files/5558730/kernel.txt)
[logcat.txt](https://github.com/HermanChen/mpp/files/5558731/logcat.txt)

Justa1986, 2020-11-18:
再缩小下范围

setprop jpege_debug 0xff

所有jpeg相关log都屏蔽掉，编码异常

这两行代码，去掉的效果，编码效果还不一样

# 效果1:
```
static MPP_RET jpege_proc_hal(void *ctx, HalEncTask *task)
{
    JpegeCtx *p = (JpegeCtx *)ctx;
    JpegeSyntax *syntax = &p->syntax;
    MppEncCfgSet *cfg = p->cfg;
    MppEncPrepCfg *prep = &cfg->prep;
    MppEncCodecCfg *codec = &cfg->codec;

    // jpege_dbg_func("enter ctx %p\n", ctx);

    syntax->width       = prep->width;
    syntax->height      = prep->height;
    syntax->hor_stride  = prep->hor_stride;
    syntax->ver_stride  = prep->ver_stride;
    syntax->format      = prep->format;
    syntax->color       = prep->color;
    syntax->quality     = codec->jpeg.quant;
    syntax->q_factor    = codec->jpeg.q_factor;
    syntax->qf_min      = codec->jpeg.qf_min;
    syntax->qf_max      = codec->jpeg.qf_max;
    syntax->qtable_y    = codec->jpeg.qtable_y;
    syntax->qtable_c    = codec->jpeg.qtable_u;

    task->valid = 1;
    task->is_intra = 1;
    task->syntax.data = syntax;
    task->syntax.number = 1;

    jpege_dbg_func("leave ctx %p\n", ctx);
    return MPP_OK;
}
```
[image]

# 测试情况2
```

static MPP_RET jpege_proc_hal(void *ctx, HalEncTask *task)
{
    JpegeCtx *p = (JpegeCtx *)ctx;
    JpegeSyntax *syntax = &p->syntax;
    MppEncCfgSet *cfg = p->cfg;
    MppEncPrepCfg *prep = &cfg->prep;
    MppEncCodecCfg *codec = &cfg->codec;

    jpege_dbg_func("enter ctx %p\n", ctx);

    syntax->width       = prep->width;
    syntax->height      = prep->height;
    syntax->hor_stride  = prep->hor_stride;
    syntax->ver_stride  = prep->ver_stride;
    syntax->format      = prep->format;
    syntax->color       = prep->color;
    syntax->quality     = codec->jpeg.quant;
    syntax->q_factor    = codec->jpeg.q_factor;
    syntax->qf_min      = codec->jpeg.qf_min;
    syntax->qf_max      = codec->jpeg.qf_max;
    syntax->qtable_y    = codec->jpeg.qtable_y;
    syntax->qtable_c    = codec->jpeg.qtable_u;

    task->valid = 1;
    task->is_intra = 1;
    task->syntax.data = syntax;
    task->syntax.number = 1;

    // jpege_dbg_func("leave ctx %p\n", ctx);
    return MPP_OK;

```

[image]

qvoid, 2020-11-19:
画质差的问题，建议单步调试跟踪下q_factor的设置。
加了log才能正常编码，这个问题有点奇怪，你们有对板子进行超频等设置吗？

closed 2020-11-23 by Justa1986