HermanChen/mpp #37: rk3288 mpp解码jpeg失败

chong, 2021-08-21:
请问kernel版本和mpp版本怎么对应的呢？ kernel 4.4.194，使用最新mpp代码解码jpeg报错，确认是版本兼容问题，这个内核heckout mpp到某个分支正常过，后来再也复现不回去了。

sudo ./test/mpi_dec_test -t 8 -n 1 -f 0   -i ~/test.jpg -w 640 -h 480  -o test.yuv
mpp[1236]: mpi_dec_utils: cmd parse result:
mpp[1236]: mpi_dec_utils: input  file name: /home/firefly/test.jpg
mpp[1236]: mpi_dec_utils: output file name: test.yuv
mpp[1236]: mpi_dec_utils: config file name:
mpp[1236]: mpi_dec_utils: width      :  640
mpp[1236]: mpi_dec_utils: height     :  480
mpp[1236]: mpi_dec_utils: type       : 8
mpp[1236]: mpi_dec_utils: debug flag : 0
mpp[1236]: mpi_dec_utils: max frames : 1
mpp[1236]: mpi_dec_test: mpi_dec_test start
mpp[1236]: mpi_dec_test: input file size 44055
mpp[1236]: mpp_rt: NOT found ion allocator
mpp[1236]: mpp_rt: found drm allocator
mpp[1236]: mpp_info: mpp version: ddcb2785 author: Chen Jinsen   2021-08-19 [jpegd]: Remove stream copy_flag on rk3568
mpp[1236]: mpi_dec_test: 0x80351e60 mpi_dec_test decoder test start w 640 h 480 type 8
mpp[1236]: vcodec_service: vcodec_service_cmd_send ioctl VPU_IOC_SET_REG failed ret -1 errno 14 Bad address
mpp[1236]: HAL_JPEGD_VDPU1: hal_jpegd_vdpu1_start send cmd failed 14
mpp[1236]: vcodec_service: vcodec_service_cmd_poll ioctl VPU_IOC_GET_REG failed ret -1 errno 110 Connection timed out
mpp[1236]: HAL_JPEGD_VDPU1: hal_jpegd_vdpu1_wait poll cmd failed 110
mpp[1236]: mpi_dec_test: 0x80351e60 decoded frame 0
mpp[1236]: mpi_dec_test: 0x80351e60 input 0 pkt output 0 frm decode 0 frames
mpp[1236]: mpi_dec_test: test success max memory 0.00 MB

kernel 日志：
[   49.073777] rk_vcodec: reg_init:1487: error: translate reg address failed, dumping regs
[   49.081867] rk-vcodec ff9a0000.vpu-service: reg[00]: 00000000
[   49.087609] rk-vcodec ff9a0000.vpu-service: reg[01]: 00000001
[   49.093401] rk-vcodec ff9a0000.vpu-service: reg[02]: fff80510
[   49.099152] rk-vcodec ff9a0000.vpu-service: reg[03]: 3000c000
[   49.104979] rk-vcodec ff9a0000.vpu-service: reg[04]: 1400f000
[   49.110765] rk-vcodec ff9a0000.vpu-service: reg[05]: 20001b7c
[   49.116508] rk-vcodec ff9a0000.vpu-service: reg[06]: 00009cff
[   49.122288] rk-vcodec ff9a0000.vpu-service: reg[07]: 0000003f
[   49.128029] rk-vcodec ff9a0000.vpu-service: reg[08]: 00000000
[   49.133796] rk-vcodec ff9a0000.vpu-service: reg[09]: 00000000
[   49.139571] rk-vcodec ff9a0000.vpu-service: reg[10]: 00000000
[   49.145313] rk-vcodec ff9a0000.vpu-service: reg[11]: 00000000
[   49.151084] rk-vcodec ff9a0000.vpu-service: reg[12]: 003c6001
[   49.156837] rk-vcodec ff9a0000.vpu-service: reg[13]: 00000000
[   49.162610] rk-vcodec ff9a0000.vpu-service: reg[14]: 00000000
[   49.168351] rk-vcodec ff9a0000.vpu-service: reg[15]: 00000000
[   49.174103] rk-vcodec ff9a0000.vpu-service: reg[16]: 02031890
[   49.179873] rk-vcodec ff9a0000.vpu-service: reg[17]: 05050304
[   49.185619] rk-vcodec ff9a0000.vpu-service: reg[18]: 00000404
[   49.191380] rk-vcodec ff9a0000.vpu-service: reg[19]: 10907d01
[   49.197126] rk-vcodec ff9a0000.vpu-service: reg[20]: 04030404
[   49.202906] rk-vcodec ff9a0000.vpu-service: reg[21]: 04040507
[   49.208662] rk-vcodec ff9a0000.vpu-service: reg[22]: 77020100
[   49.214408] rk-vcodec ff9a0000.vpu-service: reg[23]: 11111510
[   49.220171] rk-vcodec ff9a0000.vpu-service: reg[24]: 00000001
[   49.225915] rk-vcodec ff9a0000.vpu-service: reg[25]: 11111130
[   49.231674] rk-vcodec ff9a0000.vpu-service: reg[26]: 00000111
[   49.237419] rk-vcodec ff9a0000.vpu-service: reg[27]: 00000000
[   49.243180] rk-vcodec ff9a0000.vpu-service: reg[28]: 00000000
[   49.248940] rk-vcodec ff9a0000.vpu-service: reg[29]: 00000000
[   49.254683] rk-vcodec ff9a0000.vpu-service: reg[30]: 00000000
[   49.260447] rk-vcodec ff9a0000.vpu-service: reg[31]: 00000000
[   49.266191] rk-vcodec ff9a0000.vpu-service: reg[32]: 00000000
[   49.271950] rk-vcodec ff9a0000.vpu-service: reg[33]: 00000000
[   49.277711] rk-vcodec ff9a0000.vpu-service: reg[34]: 00000000
[   49.283456] rk-vcodec ff9a0000.vpu-service: reg[35]: 00000000
[   49.289215] rk-vcodec ff9a0000.vpu-service: reg[36]: 00000000
[   49.294964] rk-vcodec ff9a0000.vpu-service: reg[37]: 00000000
[   49.300727] rk-vcodec ff9a0000.vpu-service: reg[38]: 00000000
[   49.306472] rk-vcodec ff9a0000.vpu-service: reg[39]: 00000000
[   49.312230] rk-vcodec ff9a0000.vpu-service: reg[40]: 00000003
[   49.317994] rk-vcodec ff9a0000.vpu-service: reg[41]: 00000000
[... 97 lines cut ...]
[   49.885134] rk-vcodec ff9a0000.vpu-service: reg[139]: 00000000
[   49.890981] rk-vcodec ff9a0000.vpu-service: reg[140]: 00000000
[   49.896807] rk-vcodec ff9a0000.vpu-service: reg[141]: 00000000
[   49.902645] rk-vcodec ff9a0000.vpu-service: reg[142]: 00000000
[   49.908477] rk-vcodec ff9a0000.vpu-service: reg[143]: 00000000
[   49.914321] rk-vcodec ff9a0000.vpu-service: reg[144]: 00000000
[   49.920165] rk-vcodec ff9a0000.vpu-service: reg[145]: 00000000
[   49.925991] rk-vcodec ff9a0000.vpu-service: reg[146]: 00000000
[   49.931836] rk-vcodec ff9a0000.vpu-service: reg[147]: 00000000
[   49.937662] rk-vcodec ff9a0000.vpu-service: reg[148]: 00000000
[   49.943502] rk-vcodec ff9a0000.vpu-service: reg[149]: 00000000
[   49.949331] rk-vcodec ff9a0000.vpu-service: reg[150]: 00000000
[   49.955174] rk-vcodec ff9a0000.vpu-service: reg[151]: 00000000
[   49.961014] rk-vcodec ff9a0000.vpu-service: reg[152]: 00000000
[   49.966840] rk-vcodec ff9a0000.vpu-service: reg[153]: 00000000
[   49.972690] rk-vcodec ff9a0000.vpu-service: reg[154]: 00000000
[   49.978516] rk-vcodec ff9a0000.vpu-service: reg[155]: 00000000
[   49.984360] rk-vcodec ff9a0000.vpu-service: reg[156]: 00000000
[   49.990206] rk-vcodec ff9a0000.vpu-service: reg[157]: 00000000
[   49.996032] rk-vcodec ff9a0000.vpu-service: reg[158]: 00000000
[   50.001872] rk-vcodec ff9a0000.vpu-service: reg[159]: 00000000
[   50.007701] rk-vcodec ff9a0000.vpu-service: reg[160]: 00000000
[   50.013545] rk-vcodec ff9a0000.vpu-service: reg[161]: 00000000
[   50.019385] rk-vcodec ff9a0000.vpu-service: reg[162]: 00000000
[   50.025211] rk-vcodec ff9a0000.vpu-service: reg[163]: 00000000

HermanChen, 2021-08-23:
内核日志有更完整一些么

chong, 2021-08-23:
解决了，编译加入参数-DHAVE_DRM=1

closed 2021-08-23 by chong