HermanChen/mpp #39: Encoder api

EhsanVahab, 2022-02-08:
Dear @HermanChen, thanks for your great job.
I'm going to use **mpp** in my C++ Project to encode my stream frames. 
every thing is clear and works pretty well in [mpi_enc_test.c](https://github.com/HermanChen/mpp/blob/develop/test/mpi_enc_test.c)  but I wonder which libraries and header files have to be linked and included in my project?
is there any api documentation for ?
I'm not utilizing CMake for my project and need to put everything in my makefile.

HermanChen, 2022-02-10:
The header in directory inc is just enough for common usage.
And you can use header in osal/inc to get more functions.
Document is attached.
The Makefile is also provided if you want to use mpp in linux environment.
There are debian directory for reference.

HermanChen, 2022-02-10:
[MPP_Development_Reference.pdf](https://github.com/HermanChen/mpp/files/8037208/MPP_Development_Reference.pdf)

EhsanVahab, 2022-02-10:
> [MPP_Development_Reference.pdf](https://github.com/HermanChen/mpp/files/8037208/MPP_Development_Reference.pdf)

Dear @HermanChen, thanks for your quick response.
I wonder is there any detailed document for MppApi and MppCtx and their function?
for example I have to guess the  MppApi::control's function. or spend long time to review the code to deeply understand.

HermanChen, 2022-02-11:
emmm... the document was not updated for a long time. Just check the comment in code for its usage.

EhsanVahab, 2022-02-13:
dear @HermanChen 
thanks for your comment.
I installed mpp on a virtual machine with ubuntu 18. ==>this is my programming environment.
the I wrote a simple c++ program to take advantage of MPP. every thing is ok and my code compiled and run successfully with g++.

when I move my code on **Firefly AIO-3399ProC** , it is built but I face runtime error : **_undefined symbol: mpp_spinlock_lock_**

I post my Makefile and I deeply appreciate if you make me write.
```

CFLAGS=-O3
LDFLAGS=-lrockchip_mpp -lutils -lm -lopencv_core  -lpthread -lopencv_imgproc -lopencv_highgui -lopencv_ml -lopencv_videoio -lopencv_imgcodecs
LIBDIR=-L/home/firefly/Project/mpp/build/lib -L/usr/local/lib 
INCDIR= -I/home/firefly/Project/mpp/inc -I/home/firefly/Project/mpp/utils -I/home/firefly/Project/mpp/osal/inc

default : enc

enc : enc.o 
	g++  -o $@ $^  $(INCDIR) $(LIBDIR) $(LDFLAGS) 

%.o : %.cpp
	g++  -c  $(INCDIR) $(LIBDIR) $(LDFLAGS) $(CFLAGS)  -o $@ $<

clean :
	rm -f *.o enc

```

EhsanVahab, 2022-02-13:
> dear @HermanChen thanks for your comment. I installed mpp on a virtual machine with ubuntu 18. ==>this is my programming environment. then I wrote a simple c++ program to take advantage of MPP library. every thing is ok and my code compiled and run successfully with g++.
> 
> when I move my code on **Firefly AIO-3399ProC** , it is built but I face runtime error : **_undefined symbol: mpp_spinlock_lock_**
> 
> I post my Makefile and I deeply appreciate if you make me write.
> 
> ```
> 
> CFLAGS=-O3
> LDFLAGS=-lrockchip_mpp -lutils -lm -lopencv_core  -lpthread -lopencv_imgproc -lopencv_highgui -lopencv_ml -lopencv_videoio -lopencv_imgcodecs
> LIBDIR=-L/home/firefly/Project/mpp/build/lib -L/usr/local/lib 
> INCDIR= -I/home/firefly/Project/mpp/inc -I/home/firefly/Project/mpp/utils -I/home/firefly/Project/mpp/osal/inc
> 
> 
> default : enc
> 
> 
> enc : enc.o 
> 	g++  -o $@ $^  $(INCDIR) $(LIBDIR) $(LDFLAGS) 
> 
> 
> %.o : %.cpp
> 	g++  -c  $(INCDIR) $(LIBDIR) $(LDFLAGS) $(CFLAGS)  -o $@ $<
> 
> clean :
> 	rm -f *.o enc
> ```

solved.
the root cause was the firefly ubuntu version has its own default rockchip's mpp. and it's not compatible with the latest version

EhsanVahab, 2022-02-13:
dear @HermanChen 
thanks for your great example in mpi_enc_test.c
I finally could make a c++ simple project and take advantage of Opencv.
it may be useful for others who are interested to develop IOT project with Opencv and rockchip's processors.

https://github.com/EhsanVahab/rockchip-mpp-encoder

HermanChen, 2022-02-14:
Nice work : )

closed 2022-02-14 by HermanChen