HermanChen/mpp #42: [h264e_vepu580]: Add dpb hal func and more buffer]实现也应该同时实现h264e_vepu541

jiale-gdyd, 2022-04-08:
[[h264e_vepu580]: Add dpb hal func and more buffer](https://github.com/Caesar-github/mpp/commit/f0079f49d247288631aa42e83e20f7b2d456f2f6)
这一个实现也应该在h264e_vepu541上面类似实现，不然在rv1126上update_vepu541_syntax函数会执行default条件了，就会一直打印mpp_log_f("invalid syntax type %d\n", desc->type);

HermanChen, 2022-04-08:
收到，多谢

jiale-gdyd, 2022-04-08:
[image]
[image]
请问这两处不用加上吗？

jiale-gdyd, 2022-04-08:
[image]
还有这里

jiale-gdyd, 2022-04-08:
[image]

HermanChen, 2022-04-08:
因为这个 dpb 只有 vepu580 的 h264 会用到，其他的编码器不会用到，这个是用来处理多帧多核并行编码问题的，其他器件没有这个需求

closed 2022-04-08 by jiale-gdyd