HermanChen/mpp #46: mpp_log: can not found match soc name: radxa,radxa-cm3-io rockchip,rk3566

Wazirri, 2022-07-30:
mpi_dec_mt_test h264.h264  -w 10 -h 10 -t 7
mpi_dec_mt_test: cmd parse result:
mpi_dec_mt_test: input  file name: 
mpi_dec_mt_test: output file name: 
mpi_dec_mt_test: width      :   10
mpi_dec_mt_test: height     :   10
mpi_dec_mt_test: type       : 7
mpi_dec_mt_test: debug flag : 0
mpi_dec_mt_test: mpi_dec_test start
mpi_dec_mt_test: mpi_dec_test decoder test start w 10 h 10 type 7
mpi: mpp version: 5ce1cb85 author: Caesar Wang debian: add rules for mpp with 20190626
mpp_log: can not found match soc name: radxa,radxa-cm3-io rockchip,rk3566
hal_h264d_api: Assertion vcodec_type & ((0x00000200) | (0x00000001) | (0x00000002)) failed at hal_h264d_init:104
hal_h264d_api: hal_h264d_init hard mode error, value=0
hal_h264d_api: Assertion 0 failed at hal_h264d_init:154
mpp_device: mpp_device_init failed to find device for coding 7 type 0
mpp_rt: NOT found ion allocator
mpp_rt: found drm allocator
Segmentation fault

Wazirri, 2022-07-30:
I know build with @HermanChen  files. Now I am running this command and it does not give and error or any other output.

root@radxa-cm3-io:/home/rock/mpp/build/linux/aarch64/test# ./mpi_dec_test /home/rock/2022-06-18_15-15-54.mkv 
root@radxa-cm3-io:/home/rock/mpp/build/linux/aarch64/test#

HermanChen, 2024-07-17:
The kernel driver is not properly installed.