HermanChen/mpp #47: Build mpp, and gets Segmentation Fault

Wazirri, 2022-07-30:
If I give all parameters it gives Seg fault

root@radxa-cm3-io:/home/rock# mpi_dec_test  demo.h264 -w 10 -h 10 -t 7 
Segmentation fault

if I do not give -t 7 parameters

root@radxa-cm3-io:/home/rock# mpi_dec_test 2022-06-18_15-15-54.mkv -w 10 -h 10     
root@radxa-cm3-io:/home/rock# 

It does not give any output

HermanChen, 2022-08-01:
mpi_dec_test -i demo.h264 即可

Wazirri, 2022-08-04:
Sorry, it is working now. With your MPP build.

Now it gives such error. Can you also help?

[image]

HermanChen, 2024-07-17:
The warning is about the deprecated old timeout control

HermanChen, 2024-07-17:
It is just ok

closed 2024-07-17 by HermanChen