HermanChen/mpp #48: rv1126使用rkmedia的vdec接口进行解码jpeg图片时段错误，解码h264/h265没有问题(但是有内存泄漏问题)

jiale-gdyd, 2022-08-02:
mpp[3707]: mpp_info: mpp version: 5132a494 author: Yandong Lin   2022-07-29 [mpi_enc_test]: add sei_mode env to control sei mode

mpp[3707]: mpp_soc: Assertion soc_info->vcodec_type == vcodec_type failed at MppSocService:794

[New LWP 3740]
[New LWP 3741]
[15924.175158] pgd = 4c046533
[15924.175197] [8ce25000] *pgd=2cb7c835, *pte=27098787, *ppte=27098e37
[15924.175227] CPU: 0 PID: 3740 Comm: mpp_dec_parser Tainted: G           O      4.19.111 #1
[15924.175240] Hardware name: Generic DT based system
[15924.175256] PC is at 0x96d2f140
[15924.175268] LR is at 0xa13fea50
[15924.175280] pc : [<96d2f140>]    lr : [<a13fea50>]    psr: 200d0010
[15924.175293] sp : 8c9da944  ip : 8ce25000  fp : 0000ab20
[15924.175306] r10: 8c9da9e0  r9 : 8cd0bed4  r8 : 0000a860
[15924.175345] r7 : a14e1ca8  r6 : 8ce25000  r5 : 8cd0bdc0  r4 : 0000ab20
[15924.175371] r3 : 00000000  r2 : 0000ab20  r1 : 8c9dc040  r0 : 8ce25000
[15924.175394] Flags: nzCv  IRQs on  FIQs on  Mode USER_32  ISA ARM  Segment user
[15924.175414] Control: 10c5387d  Table: 2cbf006a  DAC: 00000055
[15924.175437] CPU: 0 PID: 3740 Comm: mpp_dec_parser Tainted: G           O      4.19.111 #1
[15924.175452] Hardware name: Gene
Thread 14 "mpp_dec_parser" received signal SIGSEGV, Segmentation fault.
ic DT based syst[Switching to LWP 3740]
em
[15924.175488] [<b010f408>] (unwind_backtrace) from [<b010b96c>] (show_stack+0x10/0x14)
[15924.175515] [<b010b96c>] (show_stack) from [<b07dbda4>] (dump_stack+0x90/0xa4)
[15924.175546] [<b07dbda4>] (dump_stack) from [<b011373c>] (__do_user_fault+0x130/0x134)
[15924.175573] [<b011373c>] (__do_user_fault) from [<b0113980>] (do_page_fault+0x240/0x348)
[15924.175600] [<b0113980>] (do_page_fault) from [<b0113c18>] (do_DataAbort+0x4c/0xec)
[15924.175626] [<b0113c18>] (do_DataAbort) from [<b0101d7c>] (__dabt_usr+0x3c/0x40)
[15924.175643] Exception stack(0xdc077fb0 to 0xdc077ff8)
[15924.175664] 7fa0:                                     8ce25000 8c9dc040 0000ab20 00000000
[15924.175686] 7fc0: 0000ab20 8cd0bdc0 8ce25000 a14e1ca8 0000a860 8cd0bed4 8c9da9e0 0000ab20
[15924.175706] 7fe0: 8ce25000 8c9da944 a13fea50 96d2f140 200d0010 ffffffff
0x96d2f140 in ?? () from /lib/libc.so.6
(gdb) ========== app:[rvrpem] online ==========

(gdb) bt
#0  0x96d2f140 in ?? () from /lib/libc.so.6
#1  0xa13fea50 in jpegd_prepare () from /usr/lib/librockchip_mpp.so.1
#2  0x8cd062e0 in ?? ()
Backtrace stopped: previous frame identical to this frame (corrupt stack?)
(gdb)

jiale-gdyd, 2022-08-02:
[image]

jiale-gdyd, 2022-08-02:
图片是1440x810的图片

closed 2022-08-03 by jiale-gdyd

HermanChen, 2022-08-03:
问题解决了么？

Sherman86, 2022-08-06:
你好，问题解决了吗？请问是在哪个版本的sdk下测试的，对应的测试命令是什么？

jiale-gdyd, 2022-08-17:
我是用的rkmedia的解码接口测试的，发现不用硬件分配内存就可以，用硬件分配内存则不行

Sherman86, 2022-08-23:
请问使用的测试命令和sdk版本是什么？另外，该jpeg图片这边下载之后是1280*720的，请确认下。

jiale-gdyd, 2022-09-03:
图片是1280x720的没错，我是用的是rv1126_rv1109_sdk_v2.2.0。测试的时候我没有直接使用mpp编译出来的解码测试程序，二十使用sdk带的rkmedia中的RK_MPI_VDEC的解码接口进行图片的解码的。不过我在测试发现使用rkmedia的创建MEDIA_BUFFER时不使用硬件资源分配没有问题的

jiale-gdyd, 2022-09-03:
我现在重新拉了[hal_jpege_vepu2]: Fix memory leak on multi-core这个最新的代码进行同样的测试，好像没有问题了。已经挂机18个小时了，内存没有见涨。继续挂机测试看看