HermanChen/mpp #52: 请教下，测试RK3588 H264硬件编码出来的帧都是I帧，是什么原因呢？

Liwei-bst, 2022-12-10:
业务需要做RTP推流的功能，发现RK3568、3588系列的板子上硬件编码推流到播放端端无法被解码。
抓包H264裸流测试分析，RK3588 H264硬件编码出来的帧都是I帧，是什么原因呢，编码器需要做什么特殊配置吗？

相同的配置代码，Google的软编码器则正常。

MediaCodec配置：
```
MediaFormat mediaFormat = MediaFormat.createVideoFormat(MIME_TYPE_VIDEO, 720, 1280);
mediaFormat.setInteger(MediaFormat.KEY_BIT_RATE, 2500000);
mediaFormat.setInteger(MediaFormat.KEY_FRAME_RATE, 10);
mediaFormat.setInteger(MediaFormat.KEY_COLOR_FORMAT,
        MediaCodecInfo.CodecCapabilities.COLOR_FormatSurface);
mediaFormat.setInteger(MediaFormat.KEY_I_FRAME_INTERVAL, 10);
```

3588日志信息如下：

```
12-10 22:21:00.076   107 29610 I C2RKMpiEnc: IDR frame produced
12-10 22:21:00.084 291657 29603 W MediaCodec: mapFormat: no mediaType information
12-10 22:21:00.178   107 29610 I C2RKMpiEnc: IDR frame produced
12-10 22:21:00.181 291657 29603 W MediaCodec: mapFormat: no mediaType information
12-10 22:21:00.279   107 29610 I C2RKMpiEnc: IDR frame produced
12-10 22:21:00.282 291657 29603 W MediaCodec: mapFormat: no mediaType information
12-10 22:21:00.378   107 29610 I C2RKMpiEnc: IDR frame produced
12-10 22:21:00.381 291657 29603 W MediaCodec: mapFormat: no mediaType information
12-10 22:21:00.480   107 29610 I C2RKMpiEnc: IDR frame produced
12-10 22:21:00.483 291657 29603 W MediaCodec: mapFormat: no mediaType information
12-10 22:21:00.548 291854 29721 W System  : A resource failed to call close.
12-10 22:21:00.549 291854 29721 W System  : A resource failed to call close.
12-10 22:21:00.579   107 29610 I C2RKMpiEnc: IDR frame produced
12-10 22:21:00.582 291657 29603 W MediaCodec: mapFormat: no mediaType information
12-10 22:21:00.681   107 29610 I C2RKMpiEnc: IDR frame produced
12-10 22:21:00.683 291657 29603 W MediaCodec: mapFormat: no mediaType information
12-10 22:21:00.781   107 29610 I C2RKMpiEnc: IDR frame produced
12-10 22:21:00.784 291657 29603 W MediaCodec: mapFormat: no mediaType information
12-10 22:21:00.879   107 29610 I C2RKMpiEnc: IDR frame produced
12-10 22:21:00.881 291657 29603 W MediaCodec: mapFormat: no mediaType information
```

HermanChen, 2022-12-12:
这个和系统上层相关，不是单纯的MPP问题，到公司redmine上提问吧，

HermanChen, 2022-12-12:
KEY_FRAME_RATE 和 KEY_I_FRAME_INTERVAL 这两个参数自行修改下实验下吧

closed 2022-12-12 by HermanChen

EZreal-zhangxing, 2023-05-24:
请问楼主3588的板子进行编解码的时候都会报错，您是用的哪个分支进行转码的

> mpp[5784]: mpi_dec_test: cmd parse result:
mpp[5784]: mpi_dec_test: input  file name: /home/firefly/mpp/mpp_linux_cpp-master/res/Tennis1080p.h264
mpp[5784]: mpi_dec_test: output file name: /home/firefly/mpp/mpp_linux_cpp-master/res/Tennis1080p.yuv
mpp[5784]: mpi_dec_test: config file name:
mpp[5784]: mpi_dec_test: width      : 1920
mpp[5784]: mpi_dec_test: height     : 1080
mpp[5784]: mpi_dec_test: type       : 7
mpp[5784]: mpi_dec_test: debug flag : 0
mpp[5784]: mpi_dec_test: max frames : 5
mpp[5784]: mpi_dec_test: mpi_dec_test start
mpp[5784]: mpi_dec_test: input file size 10786528
mpp[5784]: mpp_info: mpp version: 3c5dc7fe author: sayon.chen    2020-11-23 [h265d_syntax]: Fix h265 syntax fill sps rps issue
mpp[5784]: mpi_dec_test: 0x55776107a0 mpi_dec_test decoder test start w 1920 h 1080 type 7
mpp[5784]: mpp_rt: NOT found ion allocator
mpp[5784]: mpp_rt: found drm allocator
mpp[5784]: mpp_log: can not found match soc name: rockchip,rk3588-toybrick-x10-linux rockchip,rk3588
mpp[5784]: mpi_dec_test: 0x55776107a0 decode_get_frame get info changed found
mpp[5784]: mpi_dec_test: 0x55776107a0 decoder require buffer w:h [1920:1080] stride [1920:1088] buf_size 4177920
mpp[5784]: mpi_dec_test: 0x55776107a0 decode get frame 0 err 1 discard 0
mpp[5784]: mpi_dec_test: 0x55776107a0 decode get frame 1 err 1 discard 0
mpp[5784]: mpi_dec_test: 0x55776107a0 decode get frame 2 err 1 discard 0
mpp[5784]: mpi_dec_test: 0x55776107a0 decode get frame 3 err 1 discard 0
mpp[5784]: mpi_dec_test: 0x55776107a0 decode get frame 4 err 1 discard 0
mpp[5784]: mpi_dec_test: 0x55776107a0 reach max frame number 5
mpp[5784]: mpi_dec_test: 0x55776107a0 input 432 pkt output 6 frm decode 5 frames
mpp[5784]: mpi_dec_test: test success max memory 15.94 MB

HermanChen, 2023-05-24:
2020-11-23 版本太旧了，用 develop 分支

EZreal-zhangxing, 2023-05-24:
仍然有问题
> 
May 24 17:24:33 firefly mpp[9802]: mpi_dec_utils: input file /home/firefly/mpp/mpp_linux_cpp-master/res/Tennis1080p.h264 size 10786528
May 24 17:24:33 firefly mpp[9802]: mpi_dec_utils: cmd parse result:
May 24 17:24:33 firefly mpp[9802]: mpi_dec_utils: input  file name: /home/firefly/mpp/mpp_linux_cpp-master/res/Tennis1080p.h264
May 24 17:24:33 firefly mpp[9802]: mpi_dec_utils: output file name: /home/firefly/mpp/mpp_linux_cpp-master/res/Tennis1080p.yuv
May 24 17:24:33 firefly mpp[9802]: mpi_dec_utils: width      : 1920
May 24 17:24:33 firefly mpp[9802]: mpi_dec_utils: height     : 1080
May 24 17:24:33 firefly mpp[9802]: mpi_dec_utils: type       :    7
May 24 17:24:33 firefly mpp[9802]: mpi_dec_utils: max frames :    5
May 24 17:24:33 firefly mpp[9802]: mpi_dec_test: mpi_dec_test start
May 24 17:24:33 firefly mpp[9802]: mpp_info: mpp version: ada8aed9 author: Yandong Lin   2023-03-22 [mpp_dec/enc_cfg]: fix out of bounds issue
May 24 17:24:33 firefly mpp[9802]: mpi_dec_test: 0x557b07e2e0 mpi_dec_test decoder test start w 1920 h 1080 type 7
May 24 17:24:33 firefly mpp[9802]: mpp_dma_heap: Assertion fd > 0 failed at heap_fd_open:136
May 24 17:24:33 firefly mpp[9802]: mpp_dma_heap: os_allocator_dma_heap_open open dma heap type 0 failed!
May 24 17:24:33 firefly mpp[9802]: mpp_allocator: mpp_allocator_get type 1 failed
May 24 17:24:33 firefly mpp[9802]: mpp_buffer: Assertion p->allocator failed at get_group:936
May 24 17:24:33 firefly mpp[9802]: mpp_buffer: Assertion p->alloc_api failed at get_group:937
May 24 17:24:33 firefly mpp[9802]: mpp_dma_heap: Assertion fd > 0 failed at heap_fd_open:136
May 24 17:24:33 firefly mpp[9802]: mpp_dma_heap: os_allocator_dma_heap_open open dma heap type 0 failed!
May 24 17:24:33 firefly mpp[9802]: mpp_allocator: mpp_allocator_get type 1 failed
May 24 17:24:33 firefly mpp[9802]: mpp_buffer: Assertion p->allocator failed at get_group:936
May 24 17:24:33 firefly mpp[9802]: mpp_buffer: Assertion p->alloc_api failed at get_group:937
May 24 17:25:46 firefly mpp[9921]: mpi_dec_utils: input file /home/firefly/mpp/mpp_linux_cpp-master/res/Tennis1080p.h264 size 10786528
May 24 17:25:46 firefly mpp[9921]: mpi_dec_utils: cmd parse result:
May 24 17:25:46 firefly mpp[9921]: mpi_dec_utils: input  file name: /home/firefly/mpp/mpp_linux_cpp-master/res/Tennis1080p.h264
May 24 17:25:46 firefly mpp[9921]: mpi_dec_utils: output file name: /home/firefly/mpp/mpp_linux_cpp-master/res/Tennis1080p.yuv
May 24 17:25:46 firefly mpp[9921]: mpi_dec_utils: width      : 1920
May 24 17:25:46 firefly mpp[9921]: mpi_dec_utils: height     : 1080
May 24 17:25:46 firefly mpp[9921]: mpi_dec_utils: type       :    7
May 24 17:25:46 firefly mpp[9921]: mpi_dec_utils: max frames :    5
May 24 17:25:46 firefly mpp[9921]: mpi_dec_test: mpi_dec_test start
May 24 17:25:46 firefly mpp[9921]: mpp_info: mpp version: ada8aed9 author: Yandong Lin   2023-03-22 [mpp_dec/enc_cfg]: fix out of bounds issue
May 24 17:25:46 firefly mpp[9921]: mpi_dec_test: 0x557dd1c2e0 mpi_dec_test decoder test start w 1920 h 1080 type 7
May 24 17:25:46 firefly mpp[9921]: mpp_dma_heap: Assertion fd > 0 failed at heap_fd_open:136
May 24 17:25:46 firefly mpp[9921]: mpp_dma_heap: os_allocator_dma_heap_open open dma heap type 0 failed!
May 24 17:25:46 firefly mpp[9921]: mpp_allocator: mpp_allocator_get type 1 failed
May 24 17:25:46 firefly mpp[9921]: mpp_buffer: Assertion p->allocator failed at get_group:936
May 24 17:25:46 firefly mpp[9921]: mpp_buffer: Assertion p->alloc_api failed at get_group:937
May 24 17:25:46 firefly mpp[9921]: mpp_dma_heap: Assertion fd > 0 failed at heap_fd_open:136
May 24 17:25:46 firefly mpp[9921]: mpp_dma_heap: os_allocator_dma_heap_open open dma heap type 0 failed!
May 24 17:25:46 firefly mpp[9921]: mpp_allocator: mpp_allocator_get type 1 failed
May 24 17:25:46 firefly mpp[9921]: mpp_buffer: Assertion p->allocator failed at get_group:936
May 24 17:25:46 firefly mpp[9921]: mpp_buffer: Assertion p->alloc_api failed at get_group:937

EZreal-zhangxing, 2023-05-25:
> 2020-11-23 版本太旧了，用 develop 分支

您好，这个仍然有问题。请问有解决方法吗，同时vpu_api_test 测试失败
`failed to open libvpu.so`

> May 24 17:24:33 firefly mpp[9802]: mpi_dec_utils: input file /home/firefly/mpp/mpp_linux_cpp-master/res/Tennis1080p.h264 size 10786528
May 24 17:24:33 firefly mpp[9802]: mpi_dec_utils: cmd parse result:
May 24 17:24:33 firefly mpp[9802]: mpi_dec_utils: input  file name: /home/firefly/mpp/mpp_linux_cpp-master/res/Tennis1080p.h264
May 24 17:24:33 firefly mpp[9802]: mpi_dec_utils: output file name: /home/firefly/mpp/mpp_linux_cpp-master/res/Tennis1080p.yuv
May 24 17:24:33 firefly mpp[9802]: mpi_dec_utils: width      : 1920
May 24 17:24:33 firefly mpp[9802]: mpi_dec_utils: height     : 1080
May 24 17:24:33 firefly mpp[9802]: mpi_dec_utils: type       :    7
May 24 17:24:33 firefly mpp[9802]: mpi_dec_utils: max frames :    5
May 24 17:24:33 firefly mpp[9802]: mpi_dec_test: mpi_dec_test start
May 24 17:24:33 firefly mpp[9802]: mpp_info: mpp version: ada8aed9 author: Yandong Lin   2023-03-22 [mpp_dec/enc_cfg]: fix out of bounds issue
May 24 17:24:33 firefly mpp[9802]: mpi_dec_test: 0x557b07e2e0 mpi_dec_test decoder test start w 1920 h 1080 type 7
May 24 17:24:33 firefly mpp[9802]: mpp_dma_heap: Assertion fd > 0 failed at heap_fd_open:136
May 24 17:24:33 firefly mpp[9802]: mpp_dma_heap: os_allocator_dma_heap_open open dma heap type 0 failed!
May 24 17:24:33 firefly mpp[9802]: mpp_allocator: mpp_allocator_get type 1 failed
May 24 17:24:33 firefly mpp[9802]: mpp_buffer: Assertion p->allocator failed at get_group:936
May 24 17:24:33 firefly mpp[9802]: mpp_buffer: Assertion p->alloc_api failed at get_group:937
May 24 17:24:33 firefly mpp[9802]: mpp_dma_heap: Assertion fd > 0 failed at heap_fd_open:136
May 24 17:24:33 firefly mpp[9802]: mpp_dma_heap: os_allocator_dma_heap_open open dma heap type 0 failed!
May 24 17:24:33 firefly mpp[9802]: mpp_allocator: mpp_allocator_get type 1 failed
May 24 17:24:33 firefly mpp[9802]: mpp_buffer: Assertion p->allocator failed at get_group:936
May 24 17:24:33 firefly mpp[9802]: mpp_buffer: Assertion p->alloc_api failed at get_group:937

HermanChen, 2023-05-25:
vpu_api_test 已经废弃了，不再维护，后面会删掉
mpp_dma_heap: Assertion fd > 0 failed at heap_fd_open:136
dma-heap 设备的权限问题，顺着 log 搜一下

EZreal-zhangxing, 2023-05-25:
> vpu_api_test 已经废弃了，不再维护，后面会删掉 mpp_dma_heap: Assertion fd > 0 failed at heap_fd_open:136 dma-heap 设备的权限问题，顺着 log 搜一下

感谢您的耐心解答，问题已经找到，祝您生活愉快：)