HermanChen/mpp #59: MPP can't be initialized on RK3568B2

danergo, 2023-04-22:
I want to use ffmpeg to use HW acc for decode and encode on NanoPi R5C.

Basic configuration and build of [ffmpeg](https://github.com/jjm2473/ffmpeg-rk):
```
./configure --prefix=/usr --pkg-config=pkg-config --enable-shared --enable-pthreads --enable-zlib --disable-doc --enable-debug --disable-lzma --disable-vaapi --disable-vdpau --disable-outdevs --enable-lto --enable-neon --enable-vfp --enable-gnutls --enable-libopus --enable-hardcoded-tables --enable-libmp3lame --enable-gpl --disable-libx264 --enable-version3 --enable-nonfree --enable-libdrm --enable-librga --enable-rkmpp --enable-libfdk-aac --disable-libopenh264 --disable-vaapi --disable-vdpau --disable-decoder=h264_v4l2m2m --disable-decoder=vp8_v4l2m2m --disable-decoder=mpeg2_v4l2m2m --disable-decoder=mpeg4_v4l2m2m

make -j4

sudo make install
```

ffmpeg test:

```
ffmpeg -loglevel debug -benchmark -vcodec h264_rkmpp -i Big_Buck_Bunny_1080_10s_30MB.mp4 -map 0:v:0 -f null -

ffmpeg version N-104499-g646d5566ac Copyright (c) 2000-2021 the FFmpeg developers
  built with gcc 10 (Debian 10.2.1-6)
  configuration: --prefix=/usr --pkg-config=pkg-config --enable-shared --enable-pthreads --enable-zlib --disable-doc --enable-debug --disable-lzma --disable-vaapi --disable-vdpau --disable-outdevs --enable-lto --enable-neon --enable-vfp --enable-gnutls --enable-libopus --enable-hardcoded-tables --enable-libmp3lame --enable-gpl --disable-libx264 --enable-version3 --enable-nonfree --enable-libdrm --enable-librga --enable-rkmpp --enable-libfdk-aac --disable-libopenh264 --disable-vaapi --disable-vdpau --disable-decoder=h264_v4l2m2m --disable-decoder=vp8_v4l2m2m --disable-decoder=mpeg2_v4l2m2m --disable-decoder=mpeg4_v4l2m2m
  libavutil      57.  7.100 / 57.  7.100
  libavcodec     59. 12.100 / 59. 12.100
  libavformat    59.  8.100 / 59.  8.100
  libavdevice    59.  0.101 / 59.  0.101
  libavfilter     8. 16.100 /  8. 16.100
  libswscale      6.  1.100 /  6.  1.100
  libswresample   4.  0.100 /  4.  0.100
  libpostproc    56.  0.100 / 56.  0.100
Splitting the commandline.
Reading option '-loglevel' ... matched as option 'loglevel' (set logging level) with argument 'debug'.
Reading option '-benchmark' ... matched as option 'benchmark' (add timings for benchmarking) with argument '1'.
Reading option '-vcodec' ... matched as option 'vcodec' (force video codec ('copy' to copy stream)) with argument 'h264_rkmpp'.
Reading option '-i' ... matched as input url with argument 'Big_Buck_Bunny_1080_10s_30MB.mp4'.
Reading option '-map' ... matched as option 'map' (set input stream mapping) with argument '0:v:0'.
Reading option '-f' ... matched as option 'f' (force format) with argument 'null'.
Reading option '-' ... matched as output url.
Finished splitting the commandline.
Parsing a group of options: global .
Applying option loglevel (set logging level) with argument debug.
Applying option benchmark (add timings for benchmarking) with argument 1.
Successfully parsed a group of options.
Parsing a group of options: input url Big_Buck_Bunny_1080_10s_30MB.mp4.
Applying option vcodec (force video codec ('copy' to copy stream)) with argument h264_rkmpp.
Successfully parsed a group of options.
Opening an input file: Big_Buck_Bunny_1080_10s_30MB.mp4.
[NULL @ 0x55ab9d72b0] Opening 'Big_Buck_Bunny_1080_10s_30MB.mp4' for reading
[file @ 0x55ab9d7720] Setting default whitelist 'file,crypto,data'
[mov,mp4,m4a,3gp,3g2,mj2 @ 0x55ab9d72b0] Format mov,mp4,m4a,3gp,3g2,mj2 probed with size=2048 and score=100
[mov,mp4,m4a,3gp,3g2,mj2 @ 0x55ab9d72b0] ISO: File Type Major Brand: isom
[mov,mp4,m4a,3gp,3g2,mj2 @ 0x55ab9d72b0] Unknown dref type 0x206c7275 size 12
[mov,mp4,m4a,3gp,3g2,mj2 @ 0x55ab9d72b0] Processing st: 0, edit list 0 - media time: 1024, duration: 153600
[mov,mp4,m4a,3gp,3g2,mj2 @ 0x55ab9d72b0] Offset DTS by 1024 to make first pts zero.
[mov,mp4,m4a,3gp,3g2,mj2 @ 0x55ab9d72b0] Setting codecpar->delay to 2 for stream st: 0
[mov,mp4,m4a,3gp,3g2,mj2 @ 0x55ab9d72b0] Before avformat_find_stream_info() pos: 4676 bytes read:32768 seeks:0 nb_streams:1
[h264 @ 0x55ab9d7dd0] nal_unit_type: 7(SPS), nal_ref_idc: 3
[h264 @ 0x55ab9d7dd0] nal_unit_type: 8(PPS), nal_ref_idc: 3
[h264 @ 0x55ab9d7dd0] nal_unit_type: 7(SPS), nal_ref_idc: 3
[h264 @ 0x55ab9d7dd0] nal_unit_type: 8(PPS), nal_ref_idc: 3
[h264 @ 0x55ab9d7dd0] nal_unit_type: 6(SEI), nal_ref_idc: 0
[h264 @ 0x55ab9d7dd0] nal_unit_type: 5(IDR), nal_ref_idc: 3
[h264 @ 0x55ab9d7dd0] Format yuv420p chosen by get_format().
[h264 @ 0x55ab9d7dd0] Reinit context to 1920x1088, pix_fmt: yuv420p
[h264 @ 0x55ab9d7dd0] no picture
[mov,mp4,m4a,3gp,3g2,mj2 @ 0x55ab9d72b0] All info found
[mov,mp4,m4a,3gp,3g2,mj2 @ 0x55ab9d72b0] After avformat_find_stream_info() pos: 1162504 bytes read:1162504 seeks:0 frames:1
Input #0, mov,mp4,m4a,3gp,3g2,mj2, from 'Big_Buck_Bunny_1080_10s_30MB.mp4':
  Metadata:
    major_brand     : isom
    minor_version   : 512
    compatible_brands: isomiso2avc1mp41
    title           : Big Buck Bunny, Sunflower version
    artist          : Blender Foundation 2008, Janus Bager Kristensen 2013
    composer        : Sacha Goedegebure
    encoder         : Lavf57.63.100
    comment         : Creative Commons Attribution 3.0 - http://bbb3d.renderfarming.net
    genre           : Animation
  Duration: 00:00:10.00, start: 0.000000, bitrate: 24563 kb/s
  Stream #0:0[0x1](und), 1, 1/15360: Video: h264 (High), 1 reference frame (avc1 / 0x31637661), yuv420p(progressive, left), 1920x1080 (1920x1088) [SAR 1:1 DAR 16:9], 0/1, 24559 kb/s, 30 fps, 30 tbr, 15360 tbn (default)
    Metadata:
      handler_name    : VideoHandler
      vendor_id       : [0][0][0][0]
Successfully opened the file.
Parsing a group of options: output url -.
Applying option map (set input stream mapping) with argument 0:v:0.
Applying option f (force format) with argument null.
Successfully parsed a group of options.
Opening an output file: -.
Successfully opened the file.
[h264_rkmpp @ 0x55ab9fcb20] Initializing RKMPP decoder.
[h264_rkmpp @ 0x55ab9fcb20] Failed to initialize MPP context (code = -1).
[h264_rkmpp @ 0x55ab9fcb20] Failed to initialize RKMPP decoder.
Stream mapping:
  Stream #0:0 -> #0:0 (h264 (h264_rkmpp) -> wrapped_avframe (native))
Error while opening decoder for input stream #0:0 : Unknown error occurred
bench: maxrss=22656kB
[AVIOContext @ 0x55ab9df7f0] Statistics: 1162504 bytes read, 0 seeks
```

Journal:

```
Apr 22 14:26:35 nanopi mpp[887]: mpp_dma_heap: Assertion fd > 0 failed at heap_fd_open:136
Apr 22 14:26:35 nanopi mpp[887]: mpp_dma_heap: os_allocator_dma_heap_open open dma heap type 0 failed!
Apr 22 14:26:35 nanopi mpp[887]: mpp_allocator: mpp_allocator_get type 1 failed
Apr 22 14:26:35 nanopi mpp[887]: mpp_buffer: Assertion p->allocator failed at get_group:936
Apr 22 14:26:35 nanopi mpp[887]: mpp_buffer: Assertion p->alloc_api failed at get_group:937
Apr 22 14:26:35 nanopi mpp[887]: vcodec_service: open vcodec_service /dev/mpp_service failed
Apr 22 14:26:35 nanopi mpp[887]: hal_h264d_api: mpp_dev_init failed ret: -1
Apr 22 14:26:35 nanopi mpp[887]: mpp_hal: mpp_hal_init hal h264d_rkdec init failed ret -1
Apr 22 14:26:35 nanopi mpp[887]: mpp_hal: mpp_hal_init could not found coding type 7
Apr 22 14:26:35 nanopi mpp[887]: mpp_dec: mpp_dec_init could not init hal
Apr 22 14:26:35 nanopi mpp[887]: mpp_time: mpp_clock_put invalid clock (nil)
Apr 22 14:26:35 nanopi mpp[887]: mpp_time: mpp_clock_put invalid clock (nil)
Apr 22 14:26:35 nanopi mpp[887]: mpp_time: mpp_clock_put invalid clock (nil)
Apr 22 14:26:35 nanopi mpp[887]: mpp_time: mpp_clock_put invalid clock (nil)
Apr 22 14:26:35 nanopi mpp[887]: mpp_time: mpp_clock_put invalid clock (nil)
Apr 22 14:26:35 nanopi mpp[887]: mpp_time: mpp_clock_put invalid clock (nil)
Apr 22 14:26:35 nanopi mpp[887]: mpp_time: mpp_clock_put invalid clock (nil)
Apr 22 14:26:35 nanopi mpp[887]: mpp_time: mpp_clock_put invalid clock (nil)
Apr 22 14:26:35 nanopi mpp[887]: mpp_time: mpp_clock_put invalid clock (nil)
Apr 22 14:26:35 nanopi mpp[887]: mpp_time: mpp_clock_put invalid clock (nil)
Apr 22 14:26:35 nanopi mpp[887]: mpp_time: mpp_clock_put invalid clock (nil)
Apr 22 14:26:35 nanopi mpp[887]: mpp: error found on mpp initialization
Apr 22 14:26:35 nanopi ffmpeg[887]: mpp_mem_pool: put_pool found 4 used buffer size 224
```

Kernel: 5.10.110

OS: Debian Bullseye

Can you please help me?

qvoid, 2023-04-23:
Could be an issue about no enough permission. Check devices under `/dev/dma_heap`  with command:
```shell
# ls -lh /dev/dma_heap/                                         
```
They are expect to be readable for normal users.

danergo, 2023-04-23:
Thanks for the tip, here is my outputs:

```
$ ls -lh /dev/dma_heap
total 0
crw------- 1 root root 251, 4 Aug  7  2022 cma
crw------- 1 root root 251, 5 Aug  7  2022 cma-uncached
crw------- 1 root root 251, 0 Aug  7  2022 system
crw------- 1 root root 251, 1 Aug  7  2022 system-dma32
crw------- 1 root root 251, 2 Aug  7  2022 system-uncached
crw------- 1 root root 251, 3 Aug  7  2022 system-uncached-dma32
```

danergo, 2023-04-23:
I tried changing their owner to `root:video`, and gave permission of `0666`, but it didn't work. I had to change it to `0777` which is quite open is this the correct way of giving permission to this feature? :)

qvoid, 2023-04-24:
The permission is control by Linux system level, by `rule.d`, not by MPP. You can Google how to change this setting. I'm not familiar with it. 
I think permission of `666` is enough, and there is no need to change owener and group to `video`.

closed 2023-08-03 by HermanChen