HermanChen/mpp #61: 最新包 aarch64

lion117, 2023-06-09:
feab136f1ae64209976d7f1d3e3536705e1dd38a

HermanChen, 2023-06-10:
这是什么情况下出现的崩溃？运行起来必现么？

lion117, 2023-06-10:
必现的。
在银河麒麟系统64位下运行就会崩溃

HermanChen, 2023-06-11:
多谢，跑下 mpp_buffer_test 看看是啥情况？
看这个补丁的修改不应该会导致问题才是，有点奇怪啊

HermanChen, 2023-06-12:
我试了下在一般的 android 平台和 linux 平台上是正常的，不用 gdb，直接跑一下看看是哪里出的问题，是 ioctl 出错还是什么问题，会不会是内核头文件不太匹配的问题
银河麒麟系统64位 用的是什么版本的内核，我们支持的内核只有 4.4/4.19/5.10 这几个版本

lion117, 2023-06-12:
明天跑一下数据给你看看

lion117, 2023-06-13:
## 运行mpp_buffer_test
> 重定向链接库   ../lib/
[image]
## 编译器版本
> gcc-linaro-7.4.1-2019.02-linux

## 系统内核版本
> 4.4.131-20181130.kylin.desktop

HermanChen, 2023-06-13:
那系统 mpp 库是什么版本的，strings libmpp.so | grep author 看看信息

lion117, 2023-06-13:
没有查看到版本信息。
同事在github的官网下载的2019的版本，修复aarch64 解码崩溃的后编译方法版本

lion117, 2023-06-13:
[mpp.zip](https://github.com/HermanChen/mpp/files/11730071/mpp.zip)
这是麒麟系统可用的mpp库

closed 2023-08-01 by lion117

HermanChen, 2023-08-03:
MPP 编译的时候，会把版本信息写入二进制文件，用 strings 的方法应该可以读出版本信息字符串来
就算是没有版本信息，也会打印出 "unknown mpp version for missing VCS info"
麻烦再确认下版本信息，19 年的库很旧了