HermanChen/mpp #63: Assistance Needed - NDA and SDK Access for Rockchip (rk3566/rk3588)

raphaelscholle, 2024-01-16:
I'm Raphael, Lead Platform Developer at OpenHD, and I'm currently deep into the development work with the rk3566 and rk3588, focusing on fine-tuning and integrating new sensors. Despite making progress on the driver side, I've hit a roadblock due to the inability to establish communication with someone from Rockchip for NDA signing and obtaining the official SDK.

Already done:

Completed drivers for rk3566 and rk3588.
Successfully retrieving RG10 data from IMX708 and IMX462 sensors.

Problem:
Lack of complete documentation and official SDK is hindering further progress.

Request for Assistance:

I'm reaching out in the hopes that someone can assist in connecting us with the right contacts at Rockchip or guide us on obtaining the necessary SDK/Documentation. At OpenHD, we are no strangers to working under NDAs and prioritize the utmost confidentiality in handling data.

How You Can Help:

Connect us with Rockchip contacts.
Provide guidance on NDA and SDK acquisition.

Why It Matters:

Your support in this matter is crucial for the success of our ongoing project. We greatly appreciate any assistance or insights you can provide!

Thank you for your time and consideration. Looking forward to positive collaborations!

Best regards,

Raphael
Lead Platform Developer
OpenHD

ChenJinsen1, 2024-01-17:
> I'm Raphael, Lead Platform Developer at OpenHD, and I'm currently deep into the development work with the rk3566 and rk3588, focusing on fine-tuning and integrating new sensors. Despite making [...]

Hi，Rockchip Senior Sales Manager PENG <pyf@rock-chips.com>，please email her and she'll help you.

raphaelscholle, 2024-01-17:
Thank you very much, I got an response :)

closed 2024-01-17 by raphaelscholle