HermanChen/mpp #68: 请教如何使用 `MPP_BUFFER_TYPE_EXT_DMA` 进行编码

Meonardo, 2024-07-02:
背景: 
* 数据源参考 [camera_source](https://github.com/HermanChen/mpp/blob/develop/utils/camera_source.c#L279);
* 使用 RGA 进行图形操作, 使用 `/dev/dma_heap/system-uncached-dma32` 分配 DMA buffer 导入到 RGA;
* 操作 RGA 后, 将上面创建的 DMA fd,  通过 `mpp_buffer_import()` 导入 `MppBuffer`;
* 参考 [mpi_enc_mt_test](https://github.com/HermanChen/mpp/blob/develop/test/mpi_enc_mt_test.cpp) 进行剩余的编码流程.

问题:
* 编码失败;
* 发现报错日志(部分):
```
...
[  968.427592] rk_iommu fdf40f00.iommu: iova = 0x00000000fe3597c0: dte_index: 0x3f8 pte_index: 0x359 page_offset: 0x7c0
[  968.427601] rk_iommu fdf40f00.iommu: mmu_dte_addr: 0x0000000004bbf000 dte@0x0000000004bbffe0: 0x2879d001 valid: 1 pte@0x000000002879dd64: 0x4bc4006 valid: 0 page@0x0000000000000000 flags: 0x0
[  968.427641] rk_iommu fdf40f00.iommu: Page fault at 0x00000000fe35ae80 of type read
[  968.427651] rk_iommu fdf40f00.iommu: iova = 0x00000000fe35ae80: dte_index: 0x3f8 pte_index: 0x35a page_offset: 0xe80
[  968.427659] rk_iommu fdf40f00.iommu: mmu_dte_addr: 0x0000000004bbf000 dte@0x0000000004bbffe0: 0x2879d001 valid: 1 pte@0x000000002879dd68: 0x4bc4006 valid: 0 page@0x0000000000000000 flags: 0x0
[  968.427693] rk_iommu fdf40f00.iommu: Page fault at 0x00000000fe3597c0 of type read
[  968.427704] rk_iommu fdf40f00.iommu: iova = 0x00000000fe3597c0: dte_index: 0x3f8 pte_index: 0x359 page_offset: 0x7c0
[  968.427713] rk_iommu fdf40f00.iommu: mmu_dte_addr: 0x0000000004bbf000 dte@0x0000000004bbffe0: 0x2879d001 valid: 1 pte@0x000000002879dd64: 0x4bc4006 valid: 0 page@0x0000000000000000 flags: 0x0
[  968.427742] rk_iommu fdf40f00.iommu: Page fault at 0x00000000fe35ae40 of type read
[  968.427757] rk_iommu fdf40f00.iommu: iova = 0x00000000fe35ae40: dte_index: 0x3f8 pte_index: 0x35a page_offset: 0xe40
[  968.427766] rk_iommu fdf40f00.iommu: mmu_dte_addr: 0x0000000004bbf000 dte@0x0000000004bbffe0: 0x2879d001 valid: 1 pte@0x000000002879dd68: 0x4bc4006 valid: 0 page@0x0000000000000000 flags: 0x0
[  968.427796] rk_iommu fdf40f00.iommu: Page fault at 0x00000000fe359800 of type read
[  968.427805] rk_iommu fdf40f00.iommu: iova = 0x00000000fe359800: dte_index: 0x3f8 pte_index: 0x359 page_offset: 0x800
[  968.427814] rk_iommu fdf40f00.iommu: mmu_dte_addr: 0x0000000004bbf000 dte@0x0000000004bbffe0: 0x2879d001 valid: 1 pte@0x000000002879dd64: 0x4bc4006 valid: 0 page@0x0000000000000000 flags: 0x0
[  968.427841] rk_iommu fdf40f00.iommu: Page fault at 0x00000000fe35a700 of type read
[  968.427850] rk_iommu fdf40f00.iommu: iova = 0x00000000fe35a700: dte_index: 0x3f8 pte_index: 0x35a page_offset: 0x700
[  968.427860] rk_iommu fdf40f00.iommu: mmu_dte_addr: 0x0000000004bbf000 dte@0x0000000004bbffe0: 0x2879d001 valid: 1 pte@0x000000002879dd68: 0x4bc4006 valid: 0 page@0x0000000000000000 flags: 0x0
```

注:
* RGA 操作是正常的, 合成画面可以正常渲染和写文件(NV12);

Meonardo, 2024-07-02:
直接使用解码后的 DMA buffer 在进行编码是正常的, 
解码是纯外部模式分配buffer, 通过 `mpp_buffer_get_fd(mpp_frame_get_buffer(frame))` 获取解码后的 DMA fd.

我的操作系统是: Android 13, MPP 版本为`mpp version: bebc9961 author: Herman Chen   2024-04-19 docs: Update 1.0.5 CHANGELOG.md`.

HermanChen, 2024-07-03:
使用 DMA_HEAP 的方式导入呢？
正常应该是可以引入的，pagefault 一般是 buffer 不够大导致的，加大下 buffer 看看

Meonardo, 2024-07-03:
@HermanChen 感谢回复.

> 使用 DMA_HEAP 的方式导入呢？

尝试过使用 `MPP_BUFFER_TYPE_DMA_HEAP`, 还是出现主楼描述的问题, 

> 正常应该是可以引入的，pagefault 一般是 buffer 不够大导致的，加大下 buffer 看看

这里指定在导入`MppBuffer` 构建 `MppBufferInfo` 时候传入 `size` 大小吗? 我在这里有设置 size 为 `1920*1088*3/2`, 我的图像格式为 `NV12`.

```c++
// 输入
enc_params.width = 1920;
enc_params.height = 1080;

...

MppBufferInfo info;
memset(&info, 0, sizeof(MppBufferInfo));
info.type = MPP_BUFFER_TYPE_EXT_DMA;
info.fd = rga_buffer->fd;
info.size = 1920*1088*3/2;
info.index = rga_buffer->fd;
ret = mpp_buffer_import(&buffer, &info);
if (ret) {
  LOGE(TAG, "mpp_buffer_import failed ret %d", ret);
  return ret;
}

ret = mpp_frame_init(&frame);
if (ret) {
  LOGE(TAG, "mpp_frame_init failed ret %d", ret);
  return ret;
}

mpp_frame_set_width(frame, enc_params.width);
mpp_frame_set_height(frame, enc_params.height);
mpp_frame_set_hor_stride(frame, enc_params.hor_stride);
mpp_frame_set_ver_stride(frame, enc_params.ver_stride);
mpp_frame_set_fmt(frame, enc_params.fmt);
mpp_frame_set_eos(frame, frm_eos);
mpp_frame_set_buffer(frame, buffer);
...
```

Meonardo, 2024-07-03:
MPP 日志:
```
MPP_ENC_SET_RC_CFG bps 6480000 [6075000 : 6885000] fps [25:25] gop 25
MPP_ENC_SET_PREP_CFG w:h [1920:1080] stride [1920:1088]
mode cbr bps [6075000:6480000:6885000] fps fix [25/1] -> fix [25/1] gop i [25] v [0]
```

HermanChen, 2024-07-03:
直接 w * h * 2 看看

Meonardo, 2024-07-03:
试过了, 还是有问题:
```
[47570.201526] rk_iommu fdf40f00.iommu: iova = 0x00000000fc859b40: dte_index: 0x3f2 pte_index: 0x59 page_offset: 0xb40
[47570.201535] rk_iommu fdf40f00.iommu: mmu_dte_addr: 0x0000000004b42000 dte@0x0000000004b42fc8: 0x658ed001 valid: 1 pte@0x00000000658ed164: 0x4b54006 valid: 0 page@0x0000000000000000 flags: 0x0
[47570.201587] rk_iommu fdf40f00.iommu: Page fault at 0x00000000fc85a2c0 of type read
[47570.201597] rk_iommu fdf40f00.iommu: iova = 0x00000000fc85a2c0: dte_index: 0x3f2 pte_index: 0x5a page_offset: 0x2c0
[47570.201607] rk_iommu fdf40f00.iommu: mmu_dte_addr: 0x0000000004b42000 dte@0x0000000004b42fc8: 0x658ed001 valid: 1 pte@0x00000000658ed168: 0x4b54006 valid: 0 page@0x0000000000000000 flags: 0x0
[47570.201644] rk_iommu fdf40f00.iommu: Page fault at 0x00000000fc85b1c0 of type read
[47570.201654] rk_iommu fdf40f00.iommu: iova = 0x00000000fc85b1c0: dte_index: 0x3f2 pte_index: 0x5b page_offset: 0x1c0
[47570.201664] rk_iommu fdf40f00.iommu: mmu_dte_addr: 0x0000000004b42000 dte@0x0000000004b42fc8: 0x658ed001 valid: 1 pte@0x00000000658ed16c: 0x4b54006 valid: 0 page@0x0000000000000000 flags: 0x0
[47570.201706] rk_iommu fdf40f00.iommu: Page fault at 0x00000000fc859b40 of type read
[47570.201716] rk_iommu fdf40f00.iommu: iova = 0x00000000fc859b40: dte_index: 0x3f2 pte_index: 0x59 page_offset: 0xb40
[47570.201725] rk_iommu fdf40f00.iommu: mmu_dte_addr: 0x0000000004b42000 dte@0x0000000004b42fc8: 0x658ed001 valid: 1 pte@0x00000000658ed164: 0x4b54006 valid: 0 page@0x0000000000000000 flags: 0x0
[47570.201757] rk_iommu fdf40f00.iommu: Page fault at 0x00000000fc85a2c0 of type read
[47570.201766] rk_iommu fdf40f00.iommu: iova = 0x00000000fc85a2c0: dte_index: 0x3f2 pte_index: 0x5a page_offset: 0x2c0
[47570.201776] rk_iommu fdf40f00.iommu: mmu_dte_addr: 0x0000000004b42000 dte@0x0000000004b42fc8: 0x658ed001 valid: 1 pte@0x00000000658ed168: 0x4b54006 valid: 0 page@0x0000000000000000 flags: 0x0
[47570.201808] rk_iommu fdf40f00.iommu: Page fault at 0x00000000fc85b1c0 of type read
[47570.201818] rk_iommu fdf40f00.iommu: iova = 0x00000000fc85b1c0: dte_index: 0x3f2 pte_index: 0x5b page_offset: 0x1c0
[47570.201828] rk_iommu fdf40f00.iommu: mmu_dte_addr: 0x0000000004b42000 dte@0x0000000004b42fc8: 0x658ed001 valid: 1 pte@0x00000000658ed16c: 0x4b54006 valid: 0 page@0x0000000000000000 flags: 0x0
```

Meonardo, 2024-07-04:
@HermanChen 能在给些建议吗? 我在redmine也提了相关问题, 但是指派的人估计没看到.

Meonardo, 2024-07-05:
buffer size 问题.

closed 2024-07-05 by Meonardo